nativescript-community/ble #1: Structure of peripheral object seems to differ between iOS and Android

lhitc, 2016-02-29:
On iOS, from what I see in the code, a "peripheral" object appears to have the structure

    {
        UUID: ...,
        name: ...,
        RSSI: ...,
        state: ...
    }

while on Android I see

    {
        type: ...,
        RSSI: ...,
        device: {
            name: ...,
            address: ...
        },
        advertisement: ...
    }

where UUID and address appear to be used synonymous in regard to consecutive actions like service discovery.

I would expect the structure to be identical regardless of OS.

EddyVerbruggen, 2016-02-29:
Yes me too - I think on Android < 5 it is though, I'll make sure it's streamlined!

EddyVerbruggen, 2016-02-29:
That's settled now; the iOS, Android <5 and Android >=5 structures now all have RSSI, UDID and name. The other bits are subject to change.

closed 2016-02-29 by EddyVerbruggen