nativescript-community/ble #2: Android 5+: Bluetooth.startScanning scanFilters don't work

lhitc, 2016-03-02:
When passing a service UUID to startScanning() for use as scan filter, no peripheral is found.
This is reproducible on at least two different BT LE capable devices with differing variants of Android 6.

it is probably due to known issues in the BT LE implementation of Android 5/6 as mentioned here: https://github.com/EddyVerbruggen/nativescript-bluetooth-demo/issues/4

In fact I had the same problem when trying this myself some time ago with NativeScript as well as with native Android code (then testing on Android 5.1), following different examples from the web, and at some point decided to rather build a workaround.

I'd be happy if some in-deep Android developer would prove me wrong.
Otherwise, to make startScanning() work as expected and fully cross platform, the only reliable solution seems to be scanning for all devices on Android 5+, doing the filtering as an afterwork in the callback's onScanResult() and from there call onDiscovery() conditionally.

The alternative would be leaving that check to the app developer. But this would violate the cross platform idea and may well result in similar bug reports in the future.

EddyVerbruggen, 2016-03-02:
Can you tell me which Android and BLE devices you tried this with and what the Service UUID is? Perhaps I can test with a similar set of devices and (perhaps) reach the same conclusions as you. Having the plugin do filtering after discovery would then make the most sense indeed.

lhitc, 2016-03-02:
In short: damn, it works now ... ;-)

Android devices:
- LG Nexus 5 with Android 6.0.1 (CM 13.0-20160209 Nightly)
- Nvidia Shield Tablet K1 with Android 6.0, build MRA58K.41524_664.1902

main testing peripheral:
blood pressure meter Medisana BW 300 connect, has service UUIDs: 1800, 1801, 180a, 180f, 1810

Originally (a few hours ago), I was scanning for service UUID 1810 (Blood Pressure) and both Android devices never found the peripheral. I did several clean builds during the test and especially no livesync.
While writing this comment I tested again with unchanged app code, but a fresh install of the plugin.
Now I see successful detections.
In both test runs I ensured that the peripheral actually has Bluetooth active when scanning.

So atm, I can't reproduce the condititions under which the problem occurs.
I did a recursive diff of the plugin (current fresh install vs. the one of the first test run) to make sure I didn't accidently break something before, but there was nothing that could have led to that.

Since this is not reproducible anymore, I suggest to close this concrete issue but to keep an eye on it.

However, while fiddling around, I found another Android issue with 128-bit-UUIDs only, but reportedly in both BLE APIs at least from 4.3 to 5.1, as the comments there suggest: https://code.google.com/p/android/issues/detail?id=59490&q=BLE&colspec=ID%20Type%20Status%20Owner%20Summary%20Stars

EddyVerbruggen, 2016-03-02:
Ah man that's unpleasant! I'll leave this open so you can add to it if something similar pops up. Thanks for your elaborate comments!

OldrichKruchna, 2016-03-15:
Hey, just one question, I am not sure if it relevant to this issue. I tried to install the bluetooth demo on two Android mobile phones (both have Android 4.4) and they were not able to find each other. Is it expected that the plugin searching only for peripherals or also it supposed to search for other BLE phones?

EddyVerbruggen, 2016-03-15:
@OldrichKruchna Good question. This plugin acts as 'Central' so it can connect to peripherals. If your phone can't act as a peripheral (which to my knowledge it can't) then that won't work.

OldrichKruchna, 2016-03-18:
Thank you for answer, I did some research and it seems that both IOS and Android support their phones to act as a peripheral, Android from version 5.0 and IOS from version 6:

[Android 5.0, an Android device can now act as a Bluetooth LE peripheral device.](http://developer.android.com/about/versions/android-5.0.html)

[As of OS X v10.9 and iOS 6, Mac and iOS devices can also function as Bluetooth low energy peripherals](https://developer.apple.com/library/ios/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/AboutCoreBluetooth/Introduction.html)

Would be pretty cool to support this functionality as well ;)

Pepan, 2016-04-05:
Hi I am testing on my Lenovo TAB2 with And. 5.0.1. I have iMac with BT keyboard and mouse but none of them (iMac 27 Retina, mouse, keyboar) are found :(.

EddyVerbruggen, 2016-06-02:
@Pepan Because either those devices are not LE, or they're already engaged in a connection which most (all?) LE devices don't support.

francorobles, 2017-09-13:
Agree with @EddyVerbruggen its either the bluetooth is not LE or its already connected to other device.

closed 2020-08-27 by farfromrefug