nativescript-community/ble #4: Should NSData be an ArrayBuffer?

vtjon, 2016-03-07:
I am starting to do more work with my BLE-to-serial adapter. When I catch the notify event, I am expecting binary data. My return type appears to be NSData. Should this be marshaled to an ArrayBuffer? I'm not sure what is the best way to handle binary data on the iOS side.

EddyVerbruggen, 2016-03-07:
It would be best to have Android and iOS return the same type so yeah that would be best.

In what way would you marshal it to work best for your usecase?

vtjon, 2016-03-07:
In my case, I believe I'm going to ultimately need it as a byte array.

EddyVerbruggen, 2016-03-07:
Would it help if you got [this data object](https://github.com/EddyVerbruggen/nativescript-bluetooth/blob/master/bluetooth.ios.js#L133) back instead?

vtjon, 2016-03-07:
Yes, I believe that will work. I am pushing the value into a buffer using the [node buffer module](https://www.npmjs.com/package/buffer) and it looks like it will work. I am converting another module that uses the node serialport package so I'm not totally sure on some of the inner workings yet.

EddyVerbruggen, 2016-03-08:
Please check the 1.1.0 version I released today (updated sample in the readme and demo app).

closed 2016-03-08 by EddyVerbruggen

vtjon, 2016-03-11:
@EddyVerbruggen Just to follow up, this new release seems to be working for my situation! Thanks for the quick work.