nativescript-community/ble #8: `/platforms/android/README.md is depricated`warning when using plugin

georgeedwards, 2016-04-14:
Here is the full warning I get when I try to run an android app which uses your plugin.

>WARNING: The file: /Users/georgeedwards/Desktop/Personal-App/node_modules/nativescript-bluetooth/platforms/android/README.md is depricated, you can read more about what will be the expected plugin structure here: https://www.nativescript.org/blog/migrating-n-android-plugins-from-version-1.7-to-2.0

I'm using v1.1.1.

vogatsiko, 2016-07-28:
Hello Eddy

Any luck with this one?

We are evaluating your very useful plugin, however cannot build with it on Android... Getting the same error as mentioned above.  Same thing running your demo app (nativescript-bluetooth-demo).  Runs well in iOS but breaks on Android.

Any advise would be greatly appreciated.

I am using the latest native script CLI / download.

Thanks

EddyVerbruggen, 2016-07-28:
Is it a warning or error?

You can simply remove the file to avoid it..

bradmartin, 2018-04-24:
Closing since NS 4x has been released and plugins have went through many changes since then.

closed 2018-04-24 by bradmartin