nativescript-community/ble #9: Incorrect Characteristics being read ?

georgeedwards, 2016-04-18:
I have a Bluetooth module which is broadcasting a custom BLE service. I have been using the LightBlue iOS app to debug. This is showing the results I expect:
[![enter image description here][2]][2]
However, when using your plugin and then your demo app, I have found it is showing a different service on the exact same service, on the same device. Why is this?

See below:
[![enter image description here][1]][1]

  [1]: http://i.stack.imgur.com/EYkcO.png
  [2]: http://i.stack.imgur.com/9g8ZW.png

EddyVerbruggen, 2016-04-18:
Don't you see the correct properties when pressing the A000 service?

georgeedwards, 2016-04-18:
Yes, it's the A004 Vs. 180D service that seems to have been picked up incorrectly?

EddyVerbruggen, 2016-04-19:
I mean: if you press the `A000` service doesn't the screen slide to the next page where the characteristics are shown? Please attach a screenshot of that page as well.

closed 2020-08-27 by farfromrefug