nativescript-community/ble #10: value of WriteOptions missing

t4deon, 2016-06-01:
in bluetooth.d.ts is the value missing in interface WriteOptions. 
Currently, it is: 
      export interface WriteOptions extends CRUDOptions {}

But it needs to be:
    export interface WriteOptions extends CRUDOptions {
      value: any;
    }

I would push the change to a new branch but I'm not authorized to do so.
It would be great if you could apply the change. Thanks in advance.

EddyVerbruggen, 2016-06-01:
Great find! That's what Pull Requests are for btw. Are you willing to create one? :)

t4deon, 2016-06-01:
I tried to create a new branch but I don't have permission for that. So creating a pull request will not work as both branches would be 'master', right? But if I could, I would :)

closed 2016-06-01 by t4deon

tkaarrela, 2016-10-05:
Has this fix been released to npm? I still need to manually do this fix in my project after tns install...

EddyVerbruggen, 2016-10-05:
@tkaarrela Ah, nope.. but it has now (1.1.3)!