nativescript-community/ble #12: Background Support

georgeedwards, 2016-06-10:
Hi Eddy,

For a project I am doing, I need to enable background BLE updates. On Android, you seem to just create a generic background service and then pass the ble updates into in - [here](https://developer.android.com/training/run-background-service/send-request.html) but for ios it seems to require authorizing the Bluetooth updates to persist in the background - [Docs](https://developer.apple.com/library/ios/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/CoreBluetoothBackgroundProcessingForIOSApps/PerformingTasksWhileYourAppIsInTheBackground.html#//apple_ref/doc/uid/TP40013257-CH7-SW1)

I had hoped to create a generic background-service plugin, but the ios approach seems to make that difficult, so I wondered if you wanted to add background support to your app? I am more than happy to help. Alternatively, I could make a separate plugin. Let me know what you think would be best.

EddyVerbruggen, 2016-06-10:
Hi!

I think that would be very useful for at least some users so I'd love to see a fork/PR that adds it. Perhaps we can then figure out how we could make this an opt-in or opt-out optional feature of this plugin.

I will help testing your code so please let me know when you have something I can take a look at. A 'WIP' PR would probably be easiest.

Cheers,
Eddy

georgeedwards, 2016-06-10:
Awesome! I'll make a start and open a PR v. soon, then we can discuss it there

georgeedwards, 2016-06-10:
Do you like Typescript? I much prefer it, particularly for plugins as I am not great at remembering the platform specific api's. As a WIP I have just started working in a separate android typescript file, the compiled version of which I am importing into the existing files. 

Judging by your other repos you don't use TS? So I am more than happy to make the compiled JS readable and move that back into the main files when I'm done, removing any ts. 

Is that OK?

EddyVerbruggen, 2016-06-11:
Hey! I'm quite a fan of TypeScript, just not that proficient with it yet. I'm currently working on a {N} app with TS and like it a lot so far. I've also added `.d.ts` definition files to all my {N} plugins.

The reason I never created a plugin in TS is that there were reports TS was performing worse than handwritten JS but that myth was busted recently. Also in the dark old days of {N} TS support was not a first class citizen so people would have had trouble using my plugins.

If you're up for some conversion stuff then I'm on board with that, I think we should still also check in the `.js` files though as ppl may not use TS so no transpilation is done for them, rendering the plugin useless if it ony contains `.ts` files.

oaguy1, 2016-12-28:
Any update on this? I am happy to help as I have an app that requires background BLE.

EddyVerbruggen, 2016-12-28:
Hey @oaguy1 any help is highly appreciated of course!

Perhaps you can take a look at [the PR George made](#13). We were a bit stuck testing the changes if I recall correctly. I would be great if we can merge the PR to upgrade this plugin to TS, and then add background support.

oaguy1, 2016-12-28:
I should have some time this weekend. I am not too strong on TS, but no
time like the present to learn!

mn-martin, 2017-02-24:
Any updates on this? I think this feature would be beneficial to {NS} as a whole. It is kind of critical to many apps.

paulpv, 2018-02-03:
If I understood the status of this then I would be willing to finish up whatever is necessary.

There is this issue and PR #13, and much of PR #13 code appears to me to be in the master branch, so I am confused about what is missing.
And, as the OP in this issue mentions, on Android a Service is necessary in order to maintain long running background scans, but nothing in PR #13 adds the required "Service.startForeground(requestCode, notification)" to make this happen.
If there are other ways to ensure that an app isn't killed other than a "startForeground", especially related to a NativeScript, then I am all ears! :)
I am new to NativeScript, but am very familiar w/ BLE and "Foreground" background services in native Android, and less familiar w/ what all is needed in iOS.

Is the iOS code in PR #13 working/sufficient/complete?
Does PR #13 only need the Android impl completed?

@georgeedwards did a great job on this PR, and @EddyVerbruggen reviewing it, but after 1.5 years would it be worth it to start a fresh PR and drop the TS?
I have zero objections to TS itself, but that type of change should be its own separate PR, and should perhaps be its own branch.

closed 2020-08-27 by farfromrefug