nativescript-community/ble #14: Polling for bluetooth state?

dbenninger, 2016-08-08:
If the user has bluetooth switched off, do we have to poll until the user switches on bluetooth or is there any other way?

EddyVerbruggen, 2016-08-08:
Currently it is expected the user manually switches on Bluetooth and invokes the action again (whatever it was he needed Bluetooth to be on for).

However it would be quite easy to add a function that listens to the state and reports changes to the state through a callback you can pass in.

I'll leave this open so we can add this enhancement.

dbenninger, 2016-08-08:
Yes, our UI should automatically respond to bluetooth state changes. That would be a nice enhancement in this case...

coderaven, 2017-08-01:
(Nativescript Angular) - Currently we are using Angular Services and Observables to implement this. A service in the background polls for the status of the bluetooth state and the observables respond accordingly:

On Service:
```
listenToBluetoothEnabled(): Observable<boolean> {
        return new Observable(observer => {
            bluetooth.isBluetoothEnabled().then(enabled => observer.next(enabled))

            let intervalHandle = setInterval(
                () => {
                    bluetooth.isBluetoothEnabled()
                        .then(enabled => observer.next(enabled))
                }
                , 1000);

            // stop checking every second on unsubscribe
            return () => clearInterval(intervalHandle);
        });
    }
```

On Component:

```
this.bluetoothService.listenToBluetoothEnabled().subscribe(enabled => {
           // Your response here on whatever changes the bluetooth state has
        })
```

closed 2020-08-27 by farfromrefug