nativescript-community/ble #15: NativeScript Angular Support?

jiwonyune, 2016-08-17:
Hello,

I am looking for a way to use this plugin on an app built with NativeScript Angular, but it seems very tricky to me. I suppose that it's definitely possible. Has anyone done it successfully?

NirLevanon, 2016-09-12:
I think this will mean one should use the [Web Bluetooth](https://github.com/WebBluetoothCG/web-bluetooth) to implement that. Only problem would that some of the functionality of nativescript-bluetooth would have to be changed to fit the way Web Bluetooth is used in the web, so there won't be any discrepancies between web and mobile.

devCrossNet, 2016-12-05:
I use this lib in Nativescript Angular, it finds all devices and I can connect to them.

regards.

closed 2020-08-27 by farfromrefug