nativescript-community/ble #16: Bluetooth classic support

thinkoverit, 2016-08-26:
Can we have/add support for Bluetooth classic https://developer.android.com/guide/topics/connectivity/bluetooth.html from within this plugin ?

EddyVerbruggen, 2016-08-26:
This is not planned. Only LE is supported to serve the majority of usecases.

That being said of course we're open to merge any useful feature the community brings in.

closed 2016-08-26 by EddyVerbruggen

EddyVerbruggen, 2018-05-09:
@cristianmiranda That's the kind of question you'd better ask on [the forum](http://discourse.nativescript.org/).

LucasBSC, 2019-01-21:
Hello, I did not find any answer to this matter. Classic bluetooth devices are supported?

cristianmiranda, 2019-01-21:
Hi @LucasBSC , no, I ended up changing my bluetooth receiver for a BLE one which is supported. Sorry.

GPlay97, 2019-07-05:
Is there now a plugin for BT classic for NativeScript? I really need this

farfromrefug, 2019-07-06:
@GPlay97 not yet i don't think so. Any contribution would be welcome. I can share native code to do so for both platforms. Already had such a plugin for another framework for iOS and Android

GPlay97, 2019-07-07:
@farfromrefug Cool. Would be cool, if you can share it! Or maybe create plugin for nativescript? I think Bluetooth classic support is really needed :)

dimitriospafos, 2020-05-08:
@farfromrefug Could you share the native code for bluetooth classic for both android and ios, for us that don't have much experience with native code. Thank you

farfromrefug, 2020-05-08:
@dimitriospafos here is the iOS code https://github.com/Akylas/titanium_akylas_modules/blob/master/bluetooth/ios/Classes/AkylasBluetoothDeviceProxy.m. go back up a little to see the android version. Warning though I haven't touch that code for years!

dimitriospafos, 2020-05-09:
@farfromrefug Thanks will give it a try

BallisticPain, 2020-11-03:
@farfromrefug I see you have the iOS code linked in your most recent response, however, I did not see a link to Android code? Is it at the same link, but further up in the file?

I have developed a couple of NativeScript plugins in the past (small ones), but I'm just getting back into NativeScript development so I am a bit rusty. I do like that the BLE plugin doesn't use any CocoaPods or "native code". I assume this would be the recommended manner of adding BT Classic connectivity? Should this be added to this plugin, or a new one created? (Obviously, leveraging this one would be essential either way)

I have a business need to have the BT Classic implemented... so I will be continuing down this path due to that need. Unfortunately, the business use-case is Android only though they aren't opposed to it working on iOS ... I realize the time constraint would limit me to the shortest path which means I can only concern myself with Android at this juncture. That being said, the customer hasn't sent me an Android device to work with yet which will excel my need for iOS use-case. Anyways, looking for pointers and I would be happy to run with it.

Thanks!!!

farfromrefug, 2020-11-03:
@BallisticPain yes the code is there too https://github.com/Akylas/titanium_akylas_modules/blob/master/bluetooth/android/src/akylas/bluetooth/DeviceProxy.java
Let me know if you want to make it part of nativescript-community. 
You can find me on slack