nativescript-community/ble #18: Not detecting any devices

NirLevanon, 2016-09-13:
I've been testing the plugin for a while on my app for my phone that's running Android 6.0, but I haven't been able to get it to show that it can detect the bluetooth devices in my vicinity (my laptop).

- I made sure to request permission to use location
- I made sure bluetooth is enabled
- no other bluetooth devices are connected to the phone
- initiated scan, defined a callback to be called when `onDiscovered` happens
- added a `.then()` clause with a callback to see if perhaps after the scan something would come up out of th `bluetooth._connections` object. None.

:open_hands:

NirLevanon, 2016-09-13:
- Added coarse location permission to AndroidManifest.xml **and** made sure to have the permission granted through a request, using https://github.com/NathanaelA/nativescript-permissions - before I call any of the bluetooth plugin's methods. Logs confirm that the permission is granted before hand; no more "Coarse location permission has not been granted; scanning for peripherals may fail."

EddyVerbruggen, 2016-09-13:
There's also a [method in the plugin](https://github.com/EddyVerbruggen/nativescript-bluetooth/blob/master/bluetooth.d.ts#L172) to request that permission.

Can your phone find the BLE device with a different app?

NirLevanon, 2016-09-22:
Had to use an outside BLE connection app (BLE Scanner) in order to detect my laptop. Only way it works.

jimport, 2017-02-01:
I've the same issue on a OnePlus X. The weird thing is, it's working on Nexus 5X and Samsung Galaxy S6. I've tested this with the same code and the same peripheral. 

adding:
```
<uses-permission android:name="android.permission.BLUETOOTH" />
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
```
doesn't make any difference. 

Any idea how to solve this?

EddyVerbruggen, 2017-02-01:
Which Android versions are those devices running?

jimport, 2017-02-01:
Silly me, I'm sorry. The OPX and the Samsung run Android 6 and the Nexus runs Android 7.

EddyVerbruggen, 2017-02-01:
No worries and thanks for the details. So we can rule out Android version differences. Perhaps it's even down to the BLE chip in the phone. In any case I would need a OPX to debug this I'm afraid :(

jimport, 2017-02-01:
Ok, thanks for your effort. I'll have a look at it and post any news...

jimport, 2017-02-01:
Stupid me. Location was simply turned off... maybe the same happend to @NirLevanon ?

francorobles, 2017-03-29:
Hi All, Cant find device as well on iOS. Anyone experienced this?

gillestasse, 2017-07-12:
Same here. On a Galaxy S6,  I can detect using Cordova but not with nativeScript. 
I checked that Location and bluetooth are enabled.

HighSoftWare96, 2017-08-28:
Hi, the same on Huawei P9 Lite. Maybe the problem is the COARSE LOCATION permission that is not even enabled with the requestCoarseLocationPermission method.

jimport, 2017-08-30:
@gillestasse I don't have any issues on a S6 with Android 7. Which Version of `nativescript-bluetooth` do you use? You can also check `Settings > Apps > {YourApp} > Permissons` if Location permission is really enabled.

HighSoftWare96, 2017-08-30:
Yes, I was wrong: probably it's not a problem of permissions. But sometimes the plugin found the devices around sometimes not, while the bluetooth system manager had found them all. I can't explain why..

HighSoftWare96, 2017-09-05:
@EddyVerbruggen https://discourse.nativescript.org/t/nativescript-bluetooth-does-not-find-out-nearby-devices/2424

tomaszciejka, 2017-09-05:
The same happens to me. Android 711 - all permissioning is fine. When debugging the onDiscovered event never fires.

@EddyVerbruggen - let us if you have any idea how to resolve this problem or how could we help.

EddyVerbruggen, 2017-09-06:
@tomaszciejka Can you try an older version of the plugin to see if there is a problem with the latest version(s)? And what kind of BLE device are you trying to discover?

francorobles, 2017-09-11:
iOS issue solved but still cant connect on Android

tomaszciejka, 2017-09-11:
We have tried on older versions but no success.

I'm trying to connect to the other phone lumia920 and my laptop. The inbuilt bluetooth discovery list works just fine.

Equisetum, 2017-09-12:
Same here. Built in list shows all bluetooth devices (laptop, huawei P9 phone, rangefinder). Plugin sees none of them

francorobles, 2017-09-13:
Was able to figure out my issue. One more thing why is the Peripheral Id different in Android? in iOS its GUID format but in Android its like a MAC Address. @EddyVerbruggen ?

EddyVerbruggen, 2017-09-13:
@francorobles AFAIK That's according to the respective platform specs.

tomaszciejka, 2017-09-13:
@francorobles - saying 'was able to figure out my issue' - you mean the problem on Android?

francorobles, 2017-09-21:
Hi @tomaszciejka. The discovery of the BLE is now resolved, but connecting to it is still a problem. Trying to run some tests still. Fyi, im connecting to a raspberrypi.

francorobles, 2017-09-21:
I can now connect to Android @EddyVerbruggen. I was able to connect and write to the correct characteristic. 

What I did was check the length of the available services and each service's characteristic if it allows to be written to. And that's it.

tomaszciejka, 2017-09-21:
Very great news! Will you be publishing the fix or @EddyVerbruggen ?

closed 2020-08-27 by farfromrefug