nativescript-community/ble #20: "----- Invoking disc cb" explosion

NirLevanon, 2016-09-22:
Once in a while when I try to connect my device to a BLE peripheral (my laptopt, in my case) following the "Connecting" state message, I get an explosion of messages in the format `----- Invoking disc cb` which causes the app to freeze. This can also happen when I intentionally disconnect my peripheral from my device, but it's mostly bothering when it happens during the connecting state.

The message seems to originate from here:
https://github.com/EddyVerbruggen/nativescript-bluetooth/blob/81b8ae455e45fa2e2009b7b48e54aff800b1d9f6/bluetooth.android.js#L457

EddyVerbruggen, 2016-09-22:
Are you ever able to connect to a laptop? I only use real BLE hardware to test with..

NirLevanon, 2016-09-22:
Yes, I am able to connect and communicate with my laptop. I am only able to properly pass the discoverry stage 1 out 4 attempts on average.

NirLevanon, 2016-10-06:
Let me know if this helps:

```
------- _MyGattCallback.onConnectionStateChange, status: 133, new state: 0
----- invoking disc cb
----- invoking disc cb
----- invoking disc cb
----- invoking disc cb
....
```

EddyVerbruggen, 2016-10-06:
I wouldn't advise connecting to a laptop but rather a real BLE peripheral. Those disconnect callbacks are expected when a connection drops.

NirLevanon, 2016-10-06:
I'm starting to suspect a mistake on my side:
```
bluetooth.connect({
     ...
      onDisconnected: (peripheral) => {
        bluetooth.disconnect({ UUID: peripheral.UUID }).then(() => {
          console.log("Periperhal disconnected with UUID: " + peripheral.UUID);
          that.startScan();
        });
      }
....
```

I took off the call of bluetooth.disconnect - so far so good.

closed 2020-08-27 by farfromrefug