nativescript-community/ble #22: ios: isBluetoothEnabled returns false even bluetooth is on

tkaarrela, 2016-11-01:
I have had situations where isBluetoothEnabled() returns false even phone bluetooth is on.

Switching bluetooth off and on does not help. Restarting phone does not help.

Tested with version 1.1.2 & 1.1.3.

Problem has occurred with iPhone 5s & 6s.

tkaarrela, 2016-11-01:
I Updated iPhone 5s iOS version 10.1.1 and after that isBluetoothEnabled() returned true correctly. 

Is there a change that isBluetoothEnabled() does not work correctly with older iOS versions anymore?

devCrossNet, 2016-11-15:
I have the same issue with iPhone SE iOS 10.1.1.

EddyVerbruggen, 2016-11-22:
Just tested the [demo app](https://github.com/EddyVerbruggen/nativescript-bluetooth-demo) again on my iPhone 6s and the green (is available) button at the top correctly reports true/false when applicable. Can you please give that project a spin?

closed 2016-11-22 by EddyVerbruggen