nativescript-community/ble #29: Android startNotifying questions

iesrbt, 2017-01-31:
Just wondering..

About line 780:
_console.log("Characteristic " + characteristicUUID + " does not have NOTIFY or INDICATE property set");_

shouldn't it rather be:
_reject("Characteristic " + characteristicUUID + " does not have NOTIFY or INDICATE property set");
return;_

About lines 770, 771:
_reject("Set notification failed for " + characteristicUUID);
return;_

It seems some peripheral do no provide the descriptor even though they announce the characteristic is notifiable in the properties array. Does it make sense to instead do:

_bluetoothGattDescriptor=new android.bluetooth.BluetoothGattDescriptor(clientCharacteristicConfigId, android.bluetooth.BluetoothGattDescriptor.PERMISSION_WRITE);
bluetoothGattCharacteristic.addDescriptor(bluetoothGattDescriptor);
console.log("BluetoothGattDescriptor created...")_

At least that allowed me to be notified ;-)

Anyway, thanks for your plugin!

EddyVerbruggen, 2017-01-31:
That at least doesn't seem more wrong than it is now 😉 

Can you make a PR perhaps?

iesrbt, 2017-02-02:
Done.

closed 2017-02-02 by iesrbt