nativescript-community/ble #31: NullPointerException in Android when used with Workers

sam0829, 2017-02-10:
Hello, I have successfully Integrated this library in my demo project. It works as expected for connecting, writing, reading etc.

Then I tried to pull all the code into worker thread as In Android, we put BLE communication code into background thread.

Here's my implementation:
```
require("globals");
var bluetooth = require("nativescript-bluetooth");

global.onmessage = function (msg) {
if (msg.data == "START") {
  bluetooth.startScanning({
//.....properies & call back goes here
});
}
}
```
But it gives following error in Worker thread 

```
JS: Error in Bluetooth.startScanning: Error: java.lang.NullPointerException: Attempt to invoke virtual method 'int android.content.Context.checkPermission(java.lang.String, int, int)' on a null object reference
JS:     android.support.v4.content.ContextCompat.checkSelfPermission(ContextCompat.java:453)
JS:     com.tns.Runtime.WorkerGlobalOnMessageCallback(Native Method)
JS:     com.tns.Runtime.access$400(Runtime.java:33)
JS:     com.tns.Runtime$WorkerThreadHandler.handleMessage(Runtime.java:233)
JS:     android.os.Handler.dispatchMessage(Handler.java:102)
JS:     android.os.Looper.loop(Looper.java:158)
JS:     android.os.HandlerThread.run(HandlerThread.java:61)
```

So am I missing something or currently there is no support for Worker Threads?

NickIliev, 2017-03-06:
Hey @sam0829 

The thing is that the Android implementation of the plugin is using some processes from the UI thread (context, permissions, etc.) and those can not be accessed in background thread via workers. For example from API23 Android has presented some new rules for permissions and some permissions can only be explicitly requested runtime on the UI thread (via a dialog that lists the needed permissions).

sam0829, 2017-03-08:
Hello @NickIliev I understood that not everything can be moved to worker thread. So I am now trying to move specific code to worker thread like code of writing characteristics will be in background thread. I think that this approach should work as this doesn't have any UI related thing. But I am now stuck because I am not able to access the Bluetooth Object in worker thread. Is there any way I can pass an object or use an object in Worker thread?

EddyVerbruggen, 2017-03-25:
Hi @sam0829 sorry for the late reply.

See the second bullet here: https://docs.nativescript.org/core-concepts/multithreading-model#limitations

closed 2020-08-27 by farfromrefug