nativescript-community/ble #33: onNotify never calls its callback

enlight3d, 2017-03-07:
Hello, I'm writing the bluetooth communication part of an app with several other people, and I've been struggling for quite some time now so that's why I'm opening this issue here hoping someone can help !

I'm connecting to a peripheral that has a UART over BLE service, so I'm subscribing to notifications on the TX characteric of the UART service (and the logs correctly reports that notify is TRUE) but when I write values on the RX characteristic, I never receive anything while I should be receiving something (I'm sending an HEX command that works perfectly in native android code or also on cordova...).

Here is how I subscribe to notifications (notifications subscription works as I said earlier as I'm correctly seeing in the console `JS: --- notifying` - this is on Android, but works also on iOS):
```
public subscribeTXnotification = function () {
    //Subscribe to TX notifications
    console.log("----- Starting TX subscription");
    bluetooth.startNotifying({
      peripheralUUID: this.connectedPeripheral.UUID,
      serviceUUID: this.uartService,
      characteristicUUID: this.txUuid,
      onNotify: this.onData
    }).then(function() {
      console.log("----------- subscribed for notifications of TX Channel");
    });
  }
```

and the onData callback is here:
```
public onData = function (data) {
    console.log("read: ", JSON.stringify(data));
  }
```

You should note that I'm coding using TypeScript and those functions are part of a class.

Example of write call is here:
```
bluetooth.write(
  {
    peripheralUUID: that.connectedPeripheral.UUID,
    serviceUUID: that.uartService,
    characteristicUUID: that.rxUuid,
    value: '0xAA0400002955', //AA010400C555 or 55C5000401AA'
  }
).then(
  function(result) {
    console.log("Done: ", JSON.stringify(result));
  },
  function (err) {
    dialogs.alert({
      title: "Whoops!",
      message: err,
      okButtonText: "Hmmkay"
    });
  }
);
```

Also, I tried on both iOS and Android and I get the same behaviors. Write works without printing any errors in the console but I never receive anything.

enlight3d, 2017-03-09:
Okay, I understood the problem while looking at how your library works... it cannot work with 2 different characteristics at the same time, which is quite a problem for UART over BLE.

AvigarD611, 2020-08-24:
@farfromrefug I have this issue too. Doing it on iOS, with last version of this plugin. Everything works, except startNotify - I never get onNotify callback. I have tried to call .startNotify when button is clicked, then write value. Also tried to write value then .startNotify. Also tried right after .onConnected ... nothing. 

My BLE device has notifications, because I test it with other free apps in store - there I see "Notify received ...." (right after writing). 

Here is my method, that is called on button tap:

`startNotify(event) {
      var service = this.connectedDevice.services[0];

      var _that = this;

      bluetooth.startNotifying({
        peripheralUUID: _that.connectedDevice.UUID,
        serviceUUID: service.UUID,
        characteristicUUID: service.characteristics[1].UUID,
        onNotify: function (result) {
          // see the read example for how to decode ArrayBuffers
          console.log("read: " + JSON.stringify(result));
        }
      }).then(function() {
        console.log("subscribed for notifications");
      });

    }`

farfromrefug, 2020-08-25:
@maximedm indeed have it locally but not finished :s sorry. Will do that when i have the time.
Sorry!

maximedm, 2020-08-25:
@farfromrefug no problem the project is really nice the other features are working good soo keep up the good work!

farfromrefug, 2020-08-27:
@maximedm i release a beta 13. It should work my multiple char `startNotifying`. Let me know

Also i released a new plugin `@nativescript-community/ble` which will replace this one (the repo changed already) for N 7.0

closed 2020-08-27 by farfromrefug