nativescript-community/ble #34: Provide a way to toggle console logging.

Christopotamus, 2017-03-10:
Hello,

I have an app that produces a ton of bluetooth traffic (it's transferring a devices firmware over bluetooth) and on ios, the console actually completely freezes and detaches due to excessive console logging. I've tested disabling console logging in bluetooth.android.js and bluetooth.ios.js and have seen the issue disappear.

I can see how console logging could be useful, but we should provide a way to either enable or disable it. 

Would you be accepting to a pull request with some way to toggle debug output on or off?

davorpeic, 2017-08-04:
Have you tried this? https://github.com/EddyVerbruggen/nativescript-bluetooth#setcharacteristiclogging

richardlford, 2017-11-20:
I've noticed in the latest version that there are some console logs that are not protected by characteristicLogging. For example, in the Bluetooth._findCharacteristic function for ios.

bradmartin, 2018-04-24:
This will be fixed with the 2.0.0 release - small breaking changes, hence the major bump but `debug` prop will enable/disable logs.

closed 2018-04-24 by bradmartin