nativescript-community/ble #39: startScanning with serviceUUIDs doesn't return any values

sebawita, 2017-04-20:
I am using an Android 6.0.1 device (Samsung S6 edge+) and NativeScript 2.5.

I am trying to use `startScanning` function with `serviceUUIDs` parameter? However I don’t seem to be able to make it to work.
Whenever I provide `serviceUUIDs` I get 0 results

```
bluetooth.startScanning({
	seconds: 3,
	serviceUUIDs: ['ffe5'],
	// serviceUUIDs: ['0000ffe5-0000-1000-8000-00805f9b34fb'],
	onDiscovered: (peripheral: Peripheral) => {
		console.log(`UUID: ${peripheral.UUID} name: ${peripheral.name}`)
	}
})
```
I am certain that my device has a `ffe5` service, as this is the service I use for sending instructions to it.

However when I use the same `serviceUUIDs` with `web bluetooth` in a web project it seems to work. So I assume that my serviceUUIDs are correct.

I’ve tried to use both a short notation `serviceUUIDs: ['ffe5']` and a long one `serviceUUIDs: ['0000ffe5-0000-1000-8000-00805f9b34fb']` neither seems to work

EddyVerbruggen, 2017-04-21:
So I've tried a few things on iOS and Android today and can conclude a few things:

- The current implementation is the same on both platforms.
- The current implementation is per spec on both platforms.
- A peripheral can only be found if it correctly advertises the services you're searching for. My Polar H7 heartrate monitor that I'm using in the demo app advertises correctly and works fine. My TI SensorTag advertises its services incorrectly (not according to me, this is what others concluded on the web) and can't be found.

I can only assume at this point that the web bluetooth API falls back to scanning everything in case a specific service wasn't found and then filters the results based on the passed in service UUID.

This plugin could do the same, but I'm not sure that's a better approach than having the developer filter the entire result list himself.

ray007, 2020-02-24:
I seem to have the opposite problem: supplying a serviceUUID for filtering has no effect at all

Also: the `serviceUUIDs` property of a scanresults `advertisementData` is currently broken.

sebj54, 2020-03-17:
@ray007 Same problem here. I just had a look in this plugin code directly and found a working solution:

Change this:

```javascript
bluetooth.startScanning({
    serviceUUIDs: ['___YOUR_SERVICE_HERE___'],
    onDiscovered: (device) => {},
})
```

To this:
```javascript
bluetooth.startScanning({
    filters: [{
        serviceUUID: '___YOUR_SERVICE_HERE___',
    }],
    onDiscovered: (device) => {},
})
```

This syntax feels weird but it should work given this code: https://github.com/EddyVerbruggen/nativescript-bluetooth/blob/a14a8f148b8280396d3deba46047c559d9c9ee5e/src/bluetooth.android.ts#L1437

I tested it in my app and it's working fine.

@EddyVerbruggen Should the README be updated to reflect this change?

ray007, 2020-05-26:
Hmm, back in office after homeworking and testing with a real bluetooth sensor again, and suddenly things that already worked do not work anymore.

Stepping into the method `startScanning` during android debugging shows me in a logging function (bluetooth.common.js line 49). Has the API changed again?

Looking at the source on github, I also cannot find version `3.0.0-beta.10`, am I looking in the wrong place?

closed 2020-08-27 by farfromrefug