nativescript-community/ble #41: can be used with beacon?

mshanak, 2017-05-21:
can i use this plugin with beacons ( edystone/iBeacon) ?

mcgri, 2017-06-02:
I am using it with no problems.

mshanak, 2017-06-03:
how can i read the major and minor values?

seros00, 2019-01-17:
Yes, please. How can we read the Major and Minor values? Is it possible?

closed 2020-08-27 by farfromrefug

ju4nj3, 2021-07-24:
+1. Please. Someone can help and give us some light to read or discover the UUIDs beacon slots, minor or major values?
Dumped entire peripheral object onConnected event, shows another UUID different to the configured on the beacon.
Also, if I setup two slots with same UUID, same major, and different minor, can't connect to the beacon.

And, why the UUID is the MAC?

[image]

 Peripheral connected with UUID: ED:21:03:98:66:E4
 {
     "UUID": "ED:21:03:98:66:E4",
     "name": "iBKS105",
     "state": "connected",
     "services": [
         {
             "UUID": "1800",
             "characteristics": [
                 {
                     "serviceUUID": "1800",
                     "UUID": "2a00",
                     "name": "2a00",
                     "properties": {
                         "read": true,
                         "write": true,
                         "writeWithoutResponse": false,
                         "notify": false,
                         "indicate": false,
                         "broadcast": false,
                         "authenticatedSignedWrites": false,
                         "extendedProperties": false
                     },
                     "descriptors": [],
                     "permissions": null
                 },
                 {
                     "serviceUUID": "1800",

                 "UUID": "2a01",
                     "name": "2a01",
                     "properties": {
                         "read": true,
                         "write": false,
                         "writeWithoutResponse": false,
                         "notify": false,
                         "indicate": false,
                         "broadcast": false,
                         "authenticatedSignedWrites": false,
                         "extendedProperties": false
                     },
                     "descriptors": [],
                     "permissions": null
                 },
                 {
                     "serviceUUID": "1800",
                     "UUID": "2a04",
                     "name": "2a04",
                     "properties": {
                         "read": true,
                         "write": false,
                         "writeWithoutResponse": false,
                         "notify": false,
                         "indicate": false,
                         "broadcast": false,
                         "authenticatedSignedWrites": false,
                         "extendedProperties": false
                     },
                     "descriptors": [],
                     "permissions": null
                 }
             ]
[... 565 lines cut ...]
                     "UUID": "ff08",
                     "name": "ff08",
                     "properties": {
                         "read": true,
                         "write": true,
                         "writeWithoutResponse": false,
                         "notify": false,
                         "indicate": false,

         "broadcast": false,
                         "authenticatedSignedWrites": false,
                         "extendedProperties": false
                     },
                     "descriptors": [],
                     "permissions": null
                 }
             ]
         }
     ],
     "localName": "iBKS105",
     "manufacturerId": 76,
     "advertismentData": {
         "scanRecord": {}
     }
 }

ju4nj3, 2021-07-24:
+1. Please. Someone can help and give us some light to read or discover the UUIDs beacon slots, minor or major values?
Dumped entire peripheral object onConnected event, shows another UUID different to the configured on the beacon.
Also, if I setup two slots with same UUID, same major, and different minor, can't connect to the beacon.

And, why the UUID is the MAC?

[image]

`Peripheral connected with UUID: ED: 21: 03: 98: 66:E4
{
    "UUID": "ED:21:03:98:66:E4",
    "name": "iBKS105",
    "state": "connected",
    "services": [
        {
            "UUID": "1800",
            "characteristics": [
                {
                    "serviceUUID": "1800",
                    "UUID": "2a00",
                    "name": "2a00",
                    "properties": {
                        "read": true,
                        "write": true,
                        "writeWithoutResponse": false,
                        "notify": false,
                        "indicate": false,
                        "broadcast": false,
                        "authenticatedSignedWrites": false,
                        "extendedProperties": false
                    },
                    "descriptors": [],
                    "permissions": null
                },
                {
                    "serviceUUID": "1800",
                    "UUID": "2a01",
                    "name": "2a01",
                    "properties": {
                        "read": true,
                        "write": false,
                        "writeWithoutResponse": false,
                        "notify": false,
                        "indicate": false,
                        "broadcast": false,
                        "authenticatedSignedWrites": false,
                        "extendedProperties": false
                    },
                    "descriptors": [],
                    "permissions": null
                },
                {
                    "serviceUUID": "1800",
                    "UUID": "2a04",
                    "name": "2a04",
                    "properties": {
                        "read": true,
                        "write": false,
                        "writeWithoutResponse": false,
                        "notify": false,
                        "indicate": false,
                        "broadcast": false,
                        "authenticatedSignedWrites": false,
                        "extendedProperties": false
                    },
                    "descriptors": [],
                    "permissions": null
                }
            ]
        },
[... 550 lines cut ...]
        "serviceUUID": "ff00",
        "UUID": "ff08",
        "name": "ff08",
        "properties": {
            "read": true,
            "write": true,
            "writeWithoutResponse": false,
            "notify": false,
            "indicate": false,
            "broadcast": false,
            "authenticatedSignedWrites": false,
            "extendedProperties": false
        },
        "descriptors": [],
        "permissions": null
    }
]
}
],
"localName": "iBKS105",
"manufacturerId": 76,
"advertismentData": {
"scanRecord": {}
}
}`

ju4nj3, 2021-07-24:
+1. Please. Someone can help and give us some light to read or discover the UUIDs beacon slots, minor or major values?
Dumped entire peripheral object onConnected event, shows another UUID different to the configured on the beacon.
Also, if I setup two slots with same UUID, same major, and different minor, can't connect to the beacon.

And, why the UUID is the MAC?

[image]

Peripheral connected with UUID: ED: 21: 03: 98: 66:E4
{
    "UUID": "ED:21:03:98:66:E4",
    "name": "iBKS105",
    "state": "connected",
    "services": [
        {
            "UUID": "1800",
            "characteristics": [
                {
                    "serviceUUID": "1800",
                    "UUID": "2a00",
                    "name": "2a00",
                    "properties": {
                        "read": true,
                        "write": true,
                        "writeWithoutResponse": false,
                        "notify": false,
                        "indicate": false,
                        "broadcast": false,
                        "authenticatedSignedWrites": false,
                        "extendedProperties": false
                    },
                    "descriptors": [],
                    "permissions": null
                },
                {
                    "serviceUUID": "1800",
                    "UUID": "2a01",
                    "name": "2a01",
                    "properties": {
                        "read": true,
                        "write": false,
                        "writeWithoutResponse": false,
                        "notify": false,
                        "indicate": false,
                        "broadcast": false,
                        "authenticatedSignedWrites": false,
                        "extendedProperties": false
                    },
                    "descriptors": [],
                    "permissions": null
                },
                {
                    "serviceUUID": "1800",
                    "UUID": "2a04",
                    "name": "2a04",
                    "properties": {
                        "read": true,
                        "write": false,
                        "writeWithoutResponse": false,
                        "notify": false,
                        "indicate": false,
                        "broadcast": false,
                        "authenticatedSignedWrites": false,
                        "extendedProperties": false
                    },
                    "descriptors": [],
                    "permissions": null
                }
            ]
        },
[... 550 lines cut ...]
        "serviceUUID": "ff00",
        "UUID": "ff08",
        "name": "ff08",
        "properties": {
            "read": true,
            "write": true,
            "writeWithoutResponse": false,
            "notify": false,
            "indicate": false,
            "broadcast": false,
            "authenticatedSignedWrites": false,
            "extendedProperties": false
        },
        "descriptors": [],
        "permissions": null
    }
]
}
],
"localName": "iBKS105",
"manufacturerId": 76,
"advertismentData": {
"scanRecord": {}
}
}