nativescript-community/ble #43: Enable Bluetooth directly from app.

aaayushsingh, 2017-07-24:
Right now, this only supports checking if Bluetooth is enabled or not. It'd be great if it allowed turning on Bluetooth from within the application. This way, the user wouldn't have to do it manually.

EddyVerbruggen, 2017-07-24:
AFAIK that's not allowed.

aaayushsingh, 2017-07-24:
@EddyVerbruggen In android, apps can enable bluetooth. It shows up a prompt that user can choose allow or deny from. I think that's better then opening up notifications and then enabling Bluetooth.

EddyVerbruggen, 2017-07-24:
Ah interesting. Can you share an app that does that, or even better, point me at the relevant docs/code?

aaayushsingh, 2017-07-24:
@EddyVerbruggen 
`BluetoothAdapter mBluetoothAdapter = BluetoothAdapter.getDefaultAdapter(); 
    mBluetoothAdapter.enable(); `

we'll need these permissions in manifest for native android applications
`<uses-permission android:name="android.permission.BLUETOOTH"/>
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>`

solution from [this](https://stackoverflow.com/a/6276516/6753515) page.

here's a screenshot. 
[image]

Also, can you guide me on how to implement this? I'd like to do this one on my own, and then a PR maybe.

EddyVerbruggen, 2017-08-13:
PR #44 that adds this feature has just been merged. You can expect a new version on npm soon.

closed 2017-08-13 by EddyVerbruggen

aaayushsingh, 2017-08-13:
@EddyVerbruggen mind if I implement this in the Bluetooth demo app?

EddyVerbruggen, 2017-08-13:
@aaayushsingh Ha, thanks for offering but I'm working on a better implementation of this feature and also have an updated version of the demo app locally. Expecting to push it in a few hours!

EddyVerbruggen, 2017-08-13:
@aaayushsingh The demo has been updated and 1.2.0 has been published.

aaayushsingh, 2017-08-13:
So now I can boast in front of my friends about my first real PR getting merged. 

Thanks.

EddyVerbruggen, 2017-08-13:
Yes sir you can! Thanks so much, and I hope this is the first of many to any repo out there 👍 👍