nativescript-community/ble #47: Android 4.4.2 cannot connect

davorpeic, 2017-08-04:
Hi,

I'm trying to figure out why I cannot connect to my thermal printer on android (ios is working fine). I can scan and get uuid just fine, but on intent of connecting it's just get stuck and thats it, no timeouts nothing. Device is Samsung S3

I tried playing with minSdk and it is set to 21 (tried 17, 19). Just to be sure printer is working I downloaded "Bluetooth Printer" app from the store, and I see the same uuid and can make test print so I asume everything is ok with ble printer.

I started digging and I found [several](https://issuetracker.google.com/issues/36995652) [issues](https://stackoverflow.com/questions/28018722/android-could-not-connect-to-bluetooth-device-on-lollipop) related with "who gets first to connect" or something like that. My debugging took me to  `connectGatt()` method, and my printer shows blue light pointgin that maybe device is connected, but services are never returned, so sending something to write gives `gatt.getService` error.

Thanks

EddyVerbruggen, 2017-08-14:
Hi, I have no support to offer for you question, but I'm wondering if it wouldn't be easier to integrate the thermal printer manufacturer's SDK (if they have any). I'm sure that's much easier to program against than a generic Bluetooth plugin. Which printer brand is it?

davorpeic, 2017-08-14:
Hey @EddyVerbruggen 

no worries, I totally understand how support works in this cases :)

This is the thermal printer I use for testing http://www.gearbest.com/office-supplies/pp_606310.html it is a generic chinese (cheap) printer. I tried to install the test app on Redmi 4A with android 6 and it works fine, so my assumption is that connecting to this device (samsung 4.4.2) is problematic and I will drop support for it to avoid issues as connection is working on android 6.

As you already had experience with thermal printers, is there a way to send a command to connected device without listing their services first? not sure if adding new method to "print" and not to "write" would solve this issue?

EddyVerbruggen, 2017-08-14:
Hi, I only have experience with the Star line of thermal printers for which I created [a plugin](https://github.com/EddyVerbruggen/nativescript-star-printer) based on their SDK. So that's a bit higher level than what you have to do.

I think you don't need to list services before interaction, as long as you know the appropriate UUID's.

closed 2017-08-14 by EddyVerbruggen

davorpeic, 2017-08-14:
Yeah, I was checking that plugin to use it as a guide after I connect with my device. Will try few things and will let you know if I manage to do something :)
thanks!