nativescript-community/ble #50: onConnected callback not called on iOS

jimport, 2017-08-25:
Hej, I'm writing a plugin which utilizes `nativescript-bluetooth` and I ran into an issue on iOS, where the onConnected callback is not triggered after the connection to a peripheral is established. About 20 seconds after the connections is established the peripheral is disconnected. The same behavior appears with the demo app.

**iOS Versions tested:** 10.1.1, 10.3.1

Here's the console log of the demo app:
```
CONSOLE LOG file:///app/tns_modules/nativescript-bluetooth/bluetooth.js:239:16: ----- delegate centralManager:didDiscoverPeripheral: BOT @ -35
CONSOLE LOG file:///app/main-view-model.js:41:16: !!&&&&***** Clicked item with index 0
CONSOLE LOG file:///app/main-view-model.js:43:16: --- peri selected: 77571063-AFD5-4F8D-BE5E-E0A108938FF4
CONSOLE LOG file:///app/tns_modules/nativescript-bluetooth/bluetooth.js:426:20: Connecting to peripheral with UUID: 77571063-AFD5-4F8D-BE5E-E0A108938FF4
CONSOLE LOG file:///app/tns_modules/nativescript-bluetooth/bluetooth.js:264:16: ----- delegate centralManager:didConnectPeripheral: <CBPeripheral: 0x1702ea280, identifier = 77571063-AFD5-4F8D-BE5E-E0A108938FF4, name = BOT, state = connected>
CONSOLE LOG file:///app/tns_modules/nativescript-bluetooth/bluetooth.js:268:16: ----- delegate centralManager:didConnectPeripheral: cached perio: <CBPeripheral: 0x1702ea280, identifier = 77571063-AFD5-4F8D-BE5E-E0A108938FF4, name = BOT, state = connected>
CONSOLE LOG file:///app/tns_modules/nativescript-bluetooth/bluetooth.js:275:16: ----- delegate centralManager:didConnectPeripheral, let's discover service
```
The demo app was build with following versions:
```
node: 6.10.1
nativescript: 3.1.3
tns-core-modules: 3.1.1
tns-android: 3.1.1
tns-ios: 3.1.0 
```
---
In my plugin I've tested this with `nativescript-bluetooth` version 1.1.6 and 1.2.0
Here's a part of the `package.json`:
```
{
  ...
  "nativescript": {
    "platforms": {
      "android": "3.0.0",
      "ios": "3.0.0"
    }
  },
  ...
  "dependencies": {
    "nativescript-bluetooth": "1.1.6",
    "nativescript-platform": "^1.1.1",
    "nativescript-zip": "^1.3.2"
  },
  "devDependencies": {
    "tns-core-modules": "~3.0.0",
    "tns-platform-declarations": "^3.0.1",
    "typescript": "~2.2.1",
    "prompt": "~1.0.0",
    "rimraf": "~2.5.0"
  }
}
```
The app in which I'm using my plugin was build with:
```
node: 6.10.1
nativescript: 3.1.3
tns-core-modules: 3.0.1
tns-android: 3.0.0
tns-ios: 3.0.1
```
The console log is identical to the log of the demo app (except the lines of the view-model of course).

On Android everything is working fine.

Any idea how to solve this? Could version conflicts be the cause?
Thanks in advance!

mm-ns, 2017-08-25:
Bump for this. My app had been working for a while and now it has suddenly stop working and I am not sure why that is! I might have updated {N} but the onConnected callback is not triggered any longer. :( If someone has an idea in how to fix this I have a show on Tuesday where I really would need to use it. Any help is greatly appreciated.a

EddyVerbruggen, 2017-08-25:
Can you try using an older release and narrow it bit more down?

mm-ns, 2017-08-25:
older release of the library? or {N}?

jimport, 2017-08-25:
Ok, I played a bit around and found out that the peripheral in question does not advertise its service(s). Therefore the `peripheralDidDiscoverServices` delegate is not triggered.

With a peripheral which advertises its service and everything is working fine there.

mm-ns, 2017-08-25:
@jimport can you elaborate on what's missing? I'm not sure how to modify the source. This entails that we have always control over the bluetooth unit that we are interacting with, which is not the case..

jimport, 2017-08-25:
This could be an "issue" with CoreBluetooth. I've tried to connect with the LightBlue Explorer app, without success. It displays just `No services` and disconnects after a few seconds.
@mm-ns Have you tried to connect with LightBlue or something similar?

mm-ns, 2017-08-25:
no i haven't because before today my app was working just fine and I am not sure I updated anything actually. I have been running a couple of different {N} apps on my device but that's about it. :/ I found a zipped package of my project from last week (when it still was working) and if i try to run that on my device it still doesn't work. The only thing that changed from a few days ago I see im now running 3.0.3 and when i made this app i was running 3.0.1.

mm-ns, 2017-08-25:
So this is bizarre. The whole thing actually "works", if it wasn't for the fact that when I am trying to connect to my ble device from a list, it takes probably about 1 min for the callback to come back as "connected". Nothing changed on the ble device side. Any idea why this may be?

mm-ns, 2017-08-26:
Tried to revert everything to 3.0.1, but still same issues. My ble devices (arduino) are also the same code that used to work. No idea what's changed..

jimport, 2017-08-26:
It seems that this is not possible with CoreBluetooth. To discover services, they need to be advertised. So this is no issue with `nativescript-bluetooth`!

closed 2020-08-27 by farfromrefug