nativescript-community/ble #53: Use this module for peripheral devices?

sebastiendan, 2017-09-26:
Hi,

First of all, thank you for this great module!

I'm new to BLE sphere, so sorry if my wish is out of scope: is there any chance to extend this plugin to peripheral devices? I successfully exchanged data between my macbook as a peripheral (thanks to Bleno nodejs module) and my phone as a central (thanks to guess who...), now I would like to go to phone-phone interactions.

Thanks again,

Sébastien

EddyVerbruggen, 2017-09-27:
It's not planned, but leaving this open to see if anyone else:

- is interested
- wants to code it

finger563, 2017-12-11:
Definitely interested as I need this ability for nativescript to be useful, but it will take time for me to work through implementation.  Pointers / references would be appreciated :)

finger563, 2017-12-11:
I've started working on this (right now just on android) and will track some of my progress here. First step was to extend the API to wrap the calls to 
```
bluetoothAdapter.isMultipleAdvertisementSupported();
bluetoothAdapter.isOffloadedFilteringSupported();
bluetoothAdapter.isOffloadedScanBatchingSupported();
```
As a check to allow the user to query if peripheral mode is in fact supported.

closed 2020-08-27 by farfromrefug