nativescript-community/ble #54: Bluetooth enabled failing

fhackenb, 2017-10-11:
I've tried adding this plugin to an iOS app for an iPad (9.3.5) and whenever I run isBlueToothEnabled(), I get back false. I then installed the demo app https://github.com/EddyVerbruggen/nativescript-bluetooth-demo on the same device and am getting that it is not enabled there as well. My bluetooth is definitely enabled and it works as I am able to pair with a device using a different app (not one I developed).

EddyVerbruggen, 2017-10-11:
Strange.. it's this line: https://github.com/EddyVerbruggen/nativescript-bluetooth/blob/1907f7643e3f8c781dc68080ca0559db7b1358b5/bluetooth.ios.js#L311

Perhaps you can `console.log` the actual state on that device?

fhackenb, 2017-10-11:
It's being logged as undefined. Could this be an older iOS version thing?

EddyVerbruggen, 2017-10-12:
Hmm, older versions should be better than newer actually, because `CBCentralManagerStatePoweredOn` seems to be deprecated now.

I'll have to investigate and change the implementation I'm afraid. Perhaps it's best to not rely too much on that method for now..

fhackenb, 2017-10-12:
Ok. I can just have it return true for now and see if that lets me connect

fhackenb, 2017-10-17:
Returning true seemed to allow the rest of the plugin to work normally

diginikkari, 2018-04-01:
This seems to be issue also with latest ios for ipod touch 5th generation. Any idea how this could be fixed? I’m now also returning true always, but that not actual fix.

EddyVerbruggen, 2018-04-02:
It would be very useful to know if changing this bit in `bluetooth.ios.js`:

https://github.com/EddyVerbruggen/nativescript-bluetooth/blob/25ccc7067f22f2fe19a1a241900e24fb9ed5982d/bluetooth.ios.js#L340-L342

to

```js
Bluetooth._isEnabled = function () {
  return Bluetooth._state.manager.state === CBManagerState.PoweredOn;
};
```

would fix the problem. Can anyone try that perhaps?

diginikkari, 2018-04-03:
Both CBCentralManagerStatePoweredOn and CBManagerStatePoweredOn (without dot) are returning number 5. 

The issue is that for one reason Bluetooth._state.manager.state is undefined.

EddyVerbruggen, 2018-04-03:
Ah, I see. Could be it's a timing issue. Can you check that method again a few seconds after you checked it the first time?

diginikkari, 2018-04-03:
Same, still undefined.

EddyVerbruggen, 2018-04-03:
Hmm if it still works when `undefined` on your device then I may need to return `true` when `undefined` or state equals `PoweredOn`.. the latter seems to work for most devices, and the former is a fallback. I have no clue how to approach this in any other way.

diginikkari, 2018-04-03:
I think your idea is good. I just tried also with simple cordova app, and I'm getting the same undefined there also.

closed 2018-04-03 by EddyVerbruggen

dbenninger, 2018-05-07:
This change somehow didn't make it into the 1.3.1 release. The function is still:
Bluetooth._isEnabled = function () {
  return Bluetooth._state.manager.state === CBCentralManagerStatePoweredOn;
};

EddyVerbruggen, 2018-05-07:
@dbenninger You're right. I intended to do that but somehow screwed it up. Will now be part of the soon-to-be-released 2.0.0 version.

dbenninger, 2018-05-07:
Thanks, how soon?

EddyVerbruggen, 2018-05-07:
Depends on how much time I can spend in my IDE vs responding to Github issues 😘