nativescript-community/ble #57: Crashlytics issues

stevegardnersa, 2017-10-20:
I use Crashlytics and am getting the following two crash reports. I've attached the full stack trace for both. I'm not able to get to the bottom of what could cause this.

---------------------

_Fatal Exception: com.tns.NativeScriptException: 
Calling js method run failed

TypeError: Cannot read property 'stopScan' of null
File: "file:///data/data/za.co.digitlab.GFConnect/files/app/tns_modules/nativescript-bluetooth/bluetooth.js, line: 478, column: 45_

---------------------

_Fatal Exception: com.tns.NativeScriptException: 
Calling js method onServicesDiscovered failed

TypeError: Cannot read property 'onConnected' of undefined
File: "file:///data/data/za.co.digitlab.GFConnect/files/app/tns_modules/nativescript-bluetooth/bluetooth.js, line: 257, column: 17_

[za.co.digitlab.gfconnect_issue_1_crash_59E919A3031500011E5F0FFAB83B7D8C_f8b36b5ab51811e78fd856847afe9799_0_v2.txt](https://github.com/EddyVerbruggen/nativescript-bluetooth/files/1401956/za.co.digitlab.gfconnect_issue_1_crash_59E919A3031500011E5F0FFAB83B7D8C_f8b36b5ab51811e78fd856847afe9799_0_v2.txt)
[za.co.digitlab.gfconnect_issue_1_crash_59E8A30A00F6000125FDF227511FBFCE_0b68a7cab51811e78fd856847afe9799_0_v2.txt](https://github.com/EddyVerbruggen/nativescript-bluetooth/files/1401957/za.co.digitlab.gfconnect_issue_1_crash_59E8A30A00F6000125FDF227511FBFCE_0b68a7cab51811e78fd856847afe9799_0_v2.txt)

joshuamanns, 2017-11-02:
Hard to know for sure without seeing your code but I squashed a similar crash in `bluetooth.connect` by waiting until the method's Promise was fulfilled (using `then` to perform the logic rather than executing the logic in `onConnected` - which seems counterintuitive...). For me, it couldn't locate the `_wrapper` object until then. Your mileage may vary.

basilst, 2018-03-16:
I have the same problem in my project:
~~TypeError: Cannot read property 'stopScan' of null~~

I suspect the problem is on the followed code line: https://github.com/EddyVerbruggen/nativescript-bluetooth/blob/867c6dfb97a57f16d0526d2101e6aa771964fb14/bluetooth.android.js#L508
It's in a setTimeout function, therefore I guess the try-catch will not work. Will it make sense to make the ~~.stopScan~~ null save?

Strgy, 2018-11-19:
@stevegardnersa  I know its been a year since you posted this, but one of my app testers just got the same issue and I have no idea what could be causing it, all methods are properly chained with callbacks... Have you managed to pinpoint the cause of this and, wishfully, found the solution?

stevegardnersa, 2018-12-02:
@Strgy unfortunately I don't remember what we did to resolve this. I have another developer working on this piece of the codebase now and we are using a very customised version of this plugin for our specific requirements so difficult to track down. Sorry

closed 2020-08-27 by farfromrefug