nativescript-community/ble #60: bluetooth not detect devices

peppeg85, 2017-11-11:
hi, i've read the other topic about bluetooth detecting problems, but i still can't resolve it, i'm new in nativescript and mobile programming, this is my discovering function, it works until the enable  function, then the 
`onDiscovered: function (peripheral) {
                                console.log("Periperhal found with UUID: " + peripheral.UUID);
                            }`

don't return nothing,
where am i wrong?
`exports.bt = function () {
    bluetooth.isBluetoothEnabled().then(
        function (enabled) {
            console.log("Enabled? " + enabled);
            if (!enabled) {
                bluetooth.enable().then(
                    function(enabled) {
                        console.log("attivato "+ enabled);
                        bluetooth.startScanning({
                            serviceUUIDs: [],
                            seconds: 10,
                            onDiscovered: function (peripheral) {
                                console.log("Periperhal found with UUID: " + peripheral.UUID);
                            }
                        }).then(function () {
                            console.log("scanning complete ");
                            console.log("Periperhal found with UUID: " + peripheral.UUID);
                        }, function (err) {
                            console.log("error while scanning: " + err);
                        });
                    }
                );
            }
        }
    );
};`
thank you for your help

peppeg85, 2017-11-11:
sorry, now i've seen that this library is for BLE, my devices are BT standard, could this be the problem?

giordanocardillo, 2017-11-11:
+1 is this library only for BLE? and Why?

EddyVerbruggen, 2017-11-11:
Correct. This plugin is for BLE devices.

closed 2017-11-11 by EddyVerbruggen

giordanocardillo, 2017-11-11:
Is there any way I can modify this to work with Bluetooth standard devices?