nativescript-community/ble #62: writeWithoutResponse() returns too quickly on Android

kanayo, 2017-11-23:
I used your Cordova plugin to talk to a AnD BLE scale  with success.

Now porting the app over to NativeScript and using this plugin.

Ran into a problem on Android using writeWithoutResponse(). The old plugin waited for onCharacteristicWrite() to be called before calling the success callback. This plugin returns immediately.  

Now when starting another BLE call after writeWithoutResponse() returns it fails. 

I changed writeWithoutResponse() to use the same logic as write(), setting the onWritePromise, and it all works perfectly.

EddyVerbruggen, 2017-11-23:
Are you willing to share your patch by means of a PR perhaps?

kanayo, 2017-11-23:
Yes, just working on that now.

EddyVerbruggen, 2017-11-23:
That's awesome, take your time, and thanks in advance!

closed 2020-08-27 by farfromrefug