nativescript-community/ble #64: Bluetooth.startNotifying() returns immediately

kanayo, 2017-11-25:
This is similar to issue #62.

startNotifying() calls gatt.writeDescriptor() then calls resolve() right away. I think it should wait and call resolve() from onDescriptorWrite() instead.

The case I have is a Blood Pressure device that only starts sending data after you start listening to two different characteristics. This requires two calls to Bluetooth.startNotifying(), but the second one fails unless you wait for the first one to call onDescriptorWrite()

rogertthat, 2018-02-09:
This is definitely an issue, but only on Android for me (version 7.0). What I'm doing is writing bytes (using Bluetooth.write) within the promise from startNotifying. The console shows the plugin message "JS: --- notifying" before the promise is called, yet my write attempt executes the error callback, rather than the promise callback, with the error message: "Failed to write to characteristic ..."

Code:
`bluetooth.startScanning
({
    serviceUUIDs: [],
    seconds: 60,
    onDiscovered: function (peripheral)
    {
        peripheralUUID = peripheral.UUID;
        
        bluetooth.connect
        ({
            UUID: peripheralUUID,
            onConnected: function (peripheral)
            {
                bluetooth.stopScanning().then(function() {});
                
                bluetooth.startNotifying
                ({
                    peripheralUUID: peripheralUUID,
                    serviceUUID: service,
                    characteristicUUID: characteristic,
                    onNotify: function (result)
                    {
                        try
                        {
                            var data = new Uint8Array(result.value);
                            console.log('notify: '+JSON.stringify(data));
                        }
                        catch(e)
                        {
                            alert('notify value not readable: '+e.message);
                        }
                    }
                }).then(function()
                { 
                    console.log("subscribed for notifications");
                    
                    bluetooth.write
                    ({
                        peripheralUUID: peripheralUUID,
                        serviceUUID: service,
                        characteristicUUID: characteristic,
                        value: '0xa0,0x01,0xff'
                    }).then(function(result)
                    {
                        console.log('result: '+JSON.stringify(result))
                    }, function (err) {
                        alert("(1) write error: " + err);
                    });
                });
            }
        })
    }
});`

rogertthat, 2018-02-09:
@EddyVerbruggen Tagging you for an email notification, in case you don't already get them for posts in here.

rogertthat, 2018-02-12:
@EddyVerbruggen Hi Eddy, just checking in for 2 reasons: 1) Are you for hire for plugin development? My employer is interested. 2) Have you had a chance to look further into what's causing this issue?

closed 2020-08-27 by farfromrefug