nativescript-community/ble #68: Bluetooth connection remains active even after calling disconnect on android

finger563, 2017-12-11:
Happens after calling `startNotify()` on characteristics.

even after calling `stopNotify()` on all subscribed characteristics (and waiting for their promises to be resolved) and *then* calling `disconnect()`, the connection remains active.  can only be terminated by going into the settings of the phone and turning off Bluetooth.  connection even persists through app restarts (e.g. `tns run android` refreshing the app)

antfarm, 2018-02-09:
Any ideas?

finger563, 2018-02-09:
I think it's actually a device driver implementation issue, since it's not consistent across the devices I have.  I have actually gotten to where I just call `disable`, wait, then call `enable` on the Bluetooth chip which works across *most* devices - even still some retain connections.

closed 2020-08-27 by farfromrefug