nativescript-community/ble #73: Question: Does the plugin support Bluetooth classic?

DeepakArora76, 2018-02-16:
Does the plugin support Bluetooth classic? Or it for devices which support Bluetooth Low Energy (BLE)?

Is it possible to use the plugin between two paired devices (iOS-iOS, Android-Android, Android-iOS)?

To me from the source code of Android, it appears that the plugin support only BLE.

Appreciate your inputs.

EddyVerbruggen, 2018-02-16:
Yeah, and that’s also in the readme.

closed 2018-02-16 by EddyVerbruggen

DeepakArora76, 2018-02-16:
I do not find any references to Bluetooth Classic or BLE in the **Readme** document.

From the link **https://www.npmjs.com/package/nativescript-bluetooth**  what is see that:

Use when you want to
    - **scan for Bluetooth LE / Smart peripherals**,
    - **connect to those peripherals**,
    - read values of a characteristic,
    - write new values to them,
    - get notified when the value of characteristic changes.

**So, in a nutshell, Bluetooth classic is not supported by the plugin, right?**

EddyVerbruggen, 2018-02-16:
BLE only.

[image]