nativescript-community/ble #75: onNotify result

peppeg85, 2018-03-12:
hello, here i am again, i'm trying to read data sent by arduino inside the console with an hm-10 ble module, when i send something in the console i have this result:  ------- _MyGattCallback.onCharacteristicChanged
 but i can not read the characters that i've sent, where am i wrong? i've tryed with and without json stringify in the console log but nothing has changed, i've put the startNotifying inside the onConnected, is it wrong?

here is part of code:

bluetooth.connect({
            UUID: id,
            onConnected: function (peripheral) {
            console.log("onConnect")
            device.push({
                'peripheralUUID': selectedUUID,
                'serviceUUID': '0000ffe0-0000-1000-8000-00805f9b34fb',
                'characteristicUUID': '0000ffe1-0000-1000-8000-00805f9b34fb',
                'name': selectedName,
                'seriale':serial
            });
            console.log('dev :',device)
            peripheralUUID=device[0].peripheralUUID;
            serviceUUID=device[0].serviceUUID;
            characteristicUUID=device[0].characteristicUUID;
            // verifySerial(serial);
            console.log("device: ", peripheralUUID);
            console.log("deviceserviceUUID: ", serviceUUID);
            console.log("devicecharacteristicUUID: ", characteristicUUID);
            console.log("seriale: ", device[0].seriale);
            setcalendar(calendar);
            page.getViewById('bticon').style.color = 'green';
            page.getViewById('btspan').text = (selectedName + " " + id);
            page.getViewById('0').style.visibility = 'hidden';
            page.getViewById('controls').style.visibility = 'hidden';
            page.getViewById('7').style.visibility = 'visible';
            page.getViewById('1').style.visibility = 'visible';
            page.getViewById('2').style.visibility = 'visible';
            page.getViewById('3').style.visibility = 'visible';
            page.getViewById('4').style.visibility = 'visible';
            page.getViewById('5').style.visibility = 'visible';
            page.getViewById('6').style.visibility = 'visible';
            page.getViewById('8').style.visibility = 'visible';
            bluetooth.startNotifying({
                peripheralUUID: peripheralUUID,
                serviceUUID: serviceUUID,
                characteristicUUID: characteristicUUID,
                onNotify: function (result) {
                    console.log("read: ", result);
                }
            }).then(function() {
                console.log("subscribed for notifications");
            })
        },
        onDisconnected: function (peripheral) {
            console.log("Periperhal disconnected with UUID: " + selectedUUID);
        }
    });

thank you in advance

peppeg85, 2018-03-12:
sorry, i've called the function by pressing a button and the notifying works, but now i'm not able to console.log the datas since nativescript log system is tricky

peppeg85, 2018-03-17:
sorry again, is bluetooth.connect a promise? if i call the startnotify with a then after the bluetooth.connect it isnt called, if i do manually with a button it works, how can i fix this?

EddyVerbruggen, 2018-03-27:
It's a bit confusing to me that this issue is actually multiple issues in 1, but I'd like to respond to your last question: yes it's a promise, but it seemed to never resolve on Android. That will be fixed in the next version (>1.3.0).

peppeg85, 2018-03-27:
It should wait for an ack by The bt , in this moment i fixed with a timeout
of 2seconds, but its not stable

EddyVerbruggen, 2018-03-27:
That's what the `onDiscovered` callback function is for, don't use the promise for that.

peppeg85, 2018-03-27:
Ok, so the ack is not the solution. Maybe i don't use the on notify
properly?

closed 2020-08-27 by farfromrefug