nativescript-community/ble #76: write after startnotifying

peppeg85, 2018-03-19:
so, here i am again, i would try to write after the startnotifying to retrieve some data from the bluetooth interface, but if i write on the .then of startntifying i have a write error, here is the code, what is the best way to do it?

         `function setNotification(){
                         console.log("setting notification")
                         bluetooth.startNotifying({
                         peripheralUUID: periph[0],
                         serviceUUID: periph[1],
                         characteristicUUID: periph[2],
                         onNotify: function (result)
                         {
                                 console.log(result)
           
                          }
                }).then(function () {
                                  console.log("subscribed for notifications");
                                  loadEffects()
                   })
          }

    function loadEffects() {
               var action = '0xbb,0xbb,0xda'.concat(',0xf3');
              bluetooth.write({
                   peripheralUUID: periph[0],
                   serviceUUID: periph[1],
                   characteristicUUID: periph[2],
                   value: action // hex
              }).then(function (result) {
                     console.log("value written");
              }, function (err) {
                        console.log("write error: " + err);
              });
          }
`

toams69, 2018-07-14:
Got the same issue

peppeg85, 2018-07-16:
"solved" with a timeout waiting for the next release of bluetooth:

```
       bluetooth.startNotifying({
            peripheralUUID: periph[0],
            serviceUUID: periph[1],
            characteristicUUID: periph[2],
            onNotify: function (result)
            {
                console.dir(result)
            }
        }).then(function(args){
            setTimeout(function(){
                delayedFunction();
            },1500)
        })

                          function delayedFunction() {
                                 xxxxx
                              }
```

closed 2020-08-27 by farfromrefug