nativescript-community/ble #79: is it possible  permission checking outside the startScanning function?

AsrarMemon, 2018-03-27:
i am using startScanning function in worker 

But it gives following error in Worker thread
```
JS: Error in Bluetooth.startScanning: Error: java.lang.NullPointerException: Attempt to invoke virtual method 'int android.content.Context.checkPermission(java.lang.String, int, int)' on a null object reference
JS:     android.support.v4.content.ContextCompat.checkSelfPermission(ContextCompat.java:453)
JS:     com.tns.Runtime.WorkerGlobalOnMessageCallback(Native Method)
JS:     com.tns.Runtime.access$400(Runtime.java:33)
JS:     com.tns.Runtime$WorkerThreadHandler.handleMessage(Runtime.java:233)
JS:     android.os.Handler.dispatchMessage(Handler.java:102)
JS:     android.os.Looper.loop(Looper.java:158)
JS:     android.os.HandlerThread.run(HandlerThread.java:61)
```

On Android you cannot ask for permission on a non-UI thread, therefore, it will be best if a permission is checked and granted before delegating work to the worker thread.

is it possible  permission checking outside the startScanning function?

EddyVerbruggen, 2018-03-27:
Didn't the `has..` and `request..` functions [here](https://github.com/EddyVerbruggen/nativescript-bluetooth#prerequisites) work for you?

AsrarMemon, 2018-03-27:
yes i am use `has..` and `request...` function in main thread 
but  i have to use startScanning function in worker 
so i just want that can you please stop checking permission in startScanning function in nativescript-bluetooth module

EddyVerbruggen, 2018-03-27:
Ah, ok, so even if the user has granted persmission, the check in `startScanning` will fail because it's not on the main thread? Even when there's no need for a permission dialog in that case? Then I guess I can add an option to the `startScanning` method to have it not check permissions, but the default will be what it has always been of course.

AsrarMemon, 2018-03-27:
ok, great 
if you can add option for not check permissions to startScanning method

Thanks in advance :)

EddyVerbruggen, 2018-03-27:
OK, you can pass a new property to this function (see the readme and update to 1.3.1).

closed 2018-03-27 by EddyVerbruggen

AsrarMemon, 2018-03-28:
Hello,

Thank you so much............:) for add option for not check permissions
but i face other issue in new version ..i never get call back to onConnected function
my code is:
```
console.log('beforeConnect....')
bluetooth.connect({
  UUID: '04343-23445-45243-423434',
  onConnected: function (peripheral) {
  	console.log("onConnected........... ");

  	// the peripheral object now has a list of available services:
  	peripheral.services.forEach(function(service) {
  	  console.log("service found: " + JSON.stringify(service));
   });
  },
  onDisconnected: function (peripheral) {
  	console.log("Periperhal disconnected with UUID: " + peripheral.UUID);
  }
});
``` 
success fully print ''beforeConnect....'' log but never print 'onConnected........... ' means never get call back to onConnected function After update it with 1.3.1 ...

so can you please help me out .

thanks once again..:)