nativescript-community/ble #80: startScanning method with skipPermissionCheck throws error

AsrarMemon, 2018-03-28:
i use startScanning method with worker and set skipPermissionCheck option true
code : 
```
bluetooth.startScanning({
  serviceUUIDs: [],
   skipPermissionCheck: true,
  onDiscovered: function (peripheral) {
  	console.log("Periperhal found with UUID: " + peripheral.UUID);
  }
}).then(function() {
  console.log("scanning complete");
}, function (err) {
  console.log("error while scanning: " + err);
});
```
so it working properly with samsung s3 but i got error with samsung s8

error: 
```
System.err: com.tns.NativeScriptException: Cannot find object id for instance=com.tns.gen.android.bluetooth.le.ScanCallback_ftns_modules_nativescript_bluetooth_bluetooth_l93_c60__@a67fbb5
System.err:     at com.tns.Runtime.callJSMethodImpl(Runtime.java:963)
System.err:     at com.tns.Runtime.callJSMethod(Runtime.java:957)
System.err:     at com.tns.Runtime.callJSMethod(Runtime.java:941)
System.err:     at com.tns.Runtime.callJSMethod(Runtime.java:933)
System.err:     at com.tns.gen.android.bluetooth.le.ScanCallback_ftns_modules_nativescript_bluetooth_bluetooth_l93_c60__.onScanResult(ScanCallback_ftns_modules_nativescript_bluetooth_bluetooth_l93_c60__.java:25)
System.err:     at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:463)
System.err:     at android.os.Handler.handleCallback(Handler.java:751)
System.err:     at android.os.Handler.dispatchMessage(Handler.java:95)
System.err:     at android.os.Looper.loop(Looper.java:154)
System.err:     at android.app.ActivityThread.main(ActivityThread.java:6776)
System.err:     at java.lang.reflect.Method.invoke(Native Method)
System.err:     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1520)
System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1410)
ActivityManager: Process org.nativescript.PROJECTNAME (pid 24214) has died(179,1346)
ActivityManager: cleanUpApplicationRecord -- 24214
``` 

can you please help me out how to solve this error ?

EddyVerbruggen, 2018-03-28:
Do you also see this issue when *not* running in a worker?

AsrarMemon, 2018-03-28:
no without worker it's working fine

dodox87, 2018-07-13:
I have the same problem... some update? Thank you

ThomasJFR, 2018-07-30:
+1, same issue here.

patricklx, 2018-09-10:
I also had this issue, but it seems android will inoke callback only on ui thread...
E.g.:
https://stackoverflow.com/questions/29878931/android-execute-bluetoothadapter-lescancallback-on-seperate-thread

closed 2020-08-27 by farfromrefug