nativescript-community/ble #81: onConnected function never call after update 1.3.1

AsrarMemon, 2018-03-28:
i never get call back to onConnected function
my code is:

```
console.log('beforeConnect....')
bluetooth.connect({
  UUID: '04343-23445-45243-423434',
  onConnected: function (peripheral) {
  	console.log("onConnected........... ");

  	// the peripheral object now has a list of available services:
  	peripheral.services.forEach(function(service) {
  	  console.log("service found: " + JSON.stringify(service));
   });
  },
  onDisconnected: function (peripheral) {
  	console.log("Periperhal disconnected with UUID: " + peripheral.UUID);
  }
});
```

success fully print ''beforeConnect....'' log after that phone successfully connected with device but never print 'onConnected........... ' means never get call back to onConnected function After update it with 1.3.1 ...

so can you please help me out .

EddyVerbruggen, 2018-03-28:
Is this a problem in a worker only, or also when not using a worker?

AsrarMemon, 2018-03-28:
when i use startScanning with worker it is work properly with some phone

then i face this issue 

using samsung s3 phone

EddyVerbruggen, 2018-03-28:
I understand, but I'm asking if `onConnected` works outside a worker. Just to learn whether or not this is a problem with background threads vs UI threads.

AsrarMemon, 2018-03-28:
yes i understand this is a  problem with background threads but can you please help me 

why your code is not working with worker so i can understand issue 

if possible can you supports this plugin with worker ....

EddyVerbruggen, 2018-03-28:
I don't know if it can work in a worker. Perhaps those Bluetooth things need to run on the UI thread. You are the first person to try it afaik.

closed 2020-08-27 by farfromrefug