nativescript-community/ble #82: android api17: TypeError: Cannot read property 'getAdapter' of null

des-esseintes, 2018-04-02:
hello
i was using your plugin with android 5.x and it was working great, but now i've got to test it with older version of android, and it crashes w/ the following error:

```
System.err: java.lang.RuntimeException: Unable to create application com.tns.NativeScriptApplication: com.tns.NativeScriptException: 
System.err: Error calling module function 
System.err: Error calling module function 
System.err: Error calling module function 
System.err: TypeError: Cannot read property 'getAdapter' of null
System.err: File: "file:///data/data/org.nativescript.pets/files/app/tns_modules/nativescript-bluetooth/bluetooth.js, line: 90, column: 28
System.err: StackTrace: 
System.err: 	Frame: function:'', file:'file:///data/data/org.nativescript.pets/files/app/tns_modules/nativescript-bluetooth/bluetooth.js', line: 90, column: 29
System.err: 	Frame: function:'', file:'file:///data/data/org.nativescript.pets/files/app/tns_modules/nativescript-bluetooth/bluetooth.js', line: 192, column: 3
System.err: 	Frame: function:'require', file:'', line: 1, column: 266
```

could you give me an advice on how to fix this maybe? thank you!

EddyVerbruggen, 2018-04-02:
What version of Android is that?

des-esseintes, 2018-04-02:
uh totally forgot to mention, haha! it's 4.2.2 (API17), stock image inside google's emulator.
if there's a quick fix or in case im doing anything wrong, then i'd appreciate to make it work, otherwise i'll just stivk w/ higher system requirements, not a big deal

EddyVerbruggen, 2018-04-02:
Thanks. Errrr... emulator? I'm not going to investigate this unless it's reported for an actual device. Can anyone confirm this on a real device running API level 17?

des-esseintes, 2018-04-02:
I see, thank you, haha. Was very helpful.
Yes, Google has official emulator, android studio and stuff like that. This is how development is performed, dunno why it's so surprising for you. Nevermind, sorry for taking your previos time.

EddyVerbruggen, 2018-04-02:
I use emulators all the time, except for testing stuff that requires dedicated hardware (like Bluetooth). Cheers.

des-esseintes, 2018-04-03:
this question is not about hardware, it's about APIs. this is clearly stated in the error message. it seems that bt api was a bit different before android 4.3, but i might be wrong.

EddyVerbruggen, 2018-04-03:
You could be right indeed.

bradmartin, 2018-04-24:
https://developer.android.com/reference/android/bluetooth/BluetoothManager.html Android 18 for BluetoothManager

closed 2020-08-27 by farfromrefug