nativescript-community/ble #83: Scan only shows my Macbook (but none of my real devices)

d3mac123, 2018-04-05:
Hi, I am running the plugin on my iPhone (connected to my Macbook via cable so I can read the console). I have a fitness band connected to the phone but running the code only lists the Macbook and something else I cannot connect with:
```
CONSOLE LOG file:///app/dash_live-page.js:4129:22: Bluetooth Enabled? true
CONSOLE LOG file:///app/tns_modules/nativescript-bluetooth/bluetooth.js:244:16: ----- delegate centralManager:didDiscoverPeripheral: Alex\M-b\M^@\M^Ys MacBook Pro (2) @ -43
CONSOLE LOG file:///app/dash_live-page.js:4137:24: Periperhal found with UUID: D64E1857-4835-D8F6-52AE-502C27BABEC8 -43 Alex\M-b\M^@\M^Ys MacBook Pro (2) undefined
CONSOLE LOG file:///app/tns_modules/nativescript-bluetooth/bluetooth.js:244:16: ----- delegate centralManager:didDiscoverPeripheral: null @ -79
CONSOLE LOG file:///app/dash_live-page.js:4137:24: Periperhal found with UUID: 3B53F22F-BB5D-4550-A644-EEDB3416650A -79 null undefined
CONSOLE LOG file:///app/dash_live-page.js:4138:24: {"UUID":"3B53F22F-BB5D-4550-A644-EEDB3416650A","name":null,"RSSI":-74,"state":"disconnected"}
```
Here's my code:
```
bluetooth.startScanning({
        serviceUUIDs: [],
        seconds: 4,
        onDiscovered: function (peripheral) {
            console.log("Periperhal found with UUID: " + peripheral.UUID + " "+peripheral.RSSI + " "+peripheral.name + " "+peripheral.serviceUUID);
            console.log(JSON.stringify(peripheral))
        }
    }).then(function() {
        console.log("scanning complete");
    }, function (err) {
        console.log("error while scanning: " + err);
    });

    bluetooth.connect({
        UUID: '3B53F22F-BB5D-4550-A644-EEDB3416650A',
        onConnected: function (peripheral) {
            console.log("Periperhal connected with UUID: " + peripheral.UUID);
      
            // the peripheral object now has a list of available services:
            peripheral.services.forEach(function(service) {
              console.log("**** service found: " + JSON.stringify(service));
         });
        }
      });
```

Unfortunately, the bluetooth.connect() is not returning anything.
Any ideas?

EddyVerbruggen, 2018-04-05:
What fitness device is that exactly? Does is support Bluetooth LE (not 'Classic')?

d3mac123, 2018-04-05:
It's a imco fitness k4 band.

Alex Souza

EddyVerbruggen, 2018-04-06:
Is that Bluetooth LE or Bluetooth Classic?

d3mac123, 2018-04-06:
I can't find this info. I assumed it was LE due it's size/function. Same happened with my wife's FitBit band. None worked.

Alex Souza

EddyVerbruggen, 2018-04-06:
Can you pair the Fitbit in your phone settings (without providing a passcode or anything)?

d3mac123, 2018-04-06:
Yes

Alex Souza

rowdyrabouw, 2018-05-14:
I found a significant difference between iOS and Android.

On Android you can directly connect to a device with connect().
On iOS you need to do startScanning() and in the onDiscovered you can connect to the device.

closed 2020-08-27 by farfromrefug