nativescript-community/ble #84: Android can't connect to BLE 4.0

francorobles, 2018-04-06:
iOS has no trouble connecting to BLE 4.0 only on Android.

EddyVerbruggen, 2018-04-06:
Is there a device this can be tested with?

francorobles, 2018-04-06:
Hi @EddyVerbruggen, thanks for responding. Its 802.11 b/g/n, Bluetooth V4.0 + EDR.

EddyVerbruggen, 2018-04-06:
I have an S8 so the device to test the plugin on is pretty similar, but I'd also require the peripheral to test with.. is it something I can buy anywhere perhaps?

francorobles, 2018-04-07:
It is connecting to a Raspberry Pi 3B. Here is the specs https://hackaday.com/2016/02/28/introducing-the-raspberry-pi-3/

francorobles, 2018-04-13:
Hell @EddyVerbruggen . We fixed this issue by doing the following:

- Targeted 2.5.1 tns core modules.
- Added app.gradle targeting google play services 9.8.0

closed 2018-04-13 by francorobles