nativescript-community/ble #91: The state of 2.x rewrite

microwavesafe, 2018-06-15:
I was about to create a pull request, but realised the main repository is not the same version as the npm release. It looks like the 2.x branch may solve my issue (android characteristic UUID was not converted to string in notification callback), but I want to check whether anyone has used it yet and whether it is close to production ready?

EddyVerbruggen, 2018-06-15:
Struggling to find time to complete it. Well, test it, mostly. It's at ~90%.

microwavesafe, 2018-06-15:
Is the interface exactly the same? I could try installing and let you know how my testing goes?

EddyVerbruggen, 2018-06-15:
IIRC it hasn't changed - please go ahead with the version in `master`. If you have trouble getting started let me know.

microwavesafe, 2018-06-15:
I'm afraid I've fallen at the first hurdle, I can't seem to install it. I believe it is to do with my use of Vue / Webpack to compile the project, but I don't know enough about the inner workings of npm to work out what is going wrong.
```
tns plugin add nativescript-bluetooth
```
only works for pure Nativescript. So to install your 1.x package I ran
```
npm i --save nativescript-bluetooth
npm run clean
npm run watch:android 
```
I've tried using
```
npm i --save path/to/src/folder
```
which installs OK, but then I had to manually edit package-lock.json and package.json to use an absolute path otherwise the plugin wouldn't build. But then I can't import the package and webpack can't resolve the package.

I just tried creating a fork with just the npm package in it and using
```
npm i --save https://github.com/microwavesafe/nativescript-bluetooth.git
```
but this fails to install. Likely because I have not understood how the npm packaging works. Is there an easier way to install this?

microwavesafe, 2018-06-15:
I've also tried npm link from the src folder, but this lead to an error
```
Error: Cannot find module '/home/andrew/test/manual_node_packages/nativescript-bluetooth/scripts/prepare.js'
```
which doesn't seem to be part of the package.

closed 2018-09-05 by microwavesafe