nativescript-community/ble #93: Getting error on iOS

jubinpolackal, 2018-06-26:
Hi,
  I am getting the following error on iOS after updating to the latest plugin. It was working before: -

API MISUSE: Discovering services for peripheral <private> while delegate is either nil or does not implement peripheral:didDiscoverServices

Any help would be much appreciated.

Thank You.

jubinpolackal, 2018-06-27:
I got the issue fixed. Seems like the plugin was using the wrong delegate.
I made the following modifications in CBCentralManagerDelegateImpl.ts: -

import { CBPeripheralDelegateImpl } from './CBPeripheralDelegateImpl';

and in centralManagerDidConnectPeripheral method, replaced 

const delegate = CBCentralManagerDelegateImpl.new().initWithCallback(this._owner, cb);

with

const delegate = CBPeripheralDelegateImpl.new().initWithCallback(this._owner, cb);

and everything works fine now.

EddyVerbruggen, 2018-06-27:
Interesting find.. which exact version do you use? The latest from github, or 1.3.1 from npm?

chuckmitchell, 2018-06-29:
I have this issue on Master in Github. Applying Jubin's changes fixed it for me.

jubinpolackal, 2018-06-29:
@EddyVerbruggen  I used the latest version from master.

closed 2020-08-27 by farfromrefug