nativescript-community/ble #94: fork merge

farfromrefug, 2018-07-16:
HI @EddyVerbruggen I am working on a fork of your 2.x branch.
You can find it here:
https://github.com/Akylas/nativescript-bluetooth

I have changed quite a lot of things and the first and big thing is that I changed the layout of the repo. Now there is a plugin folder with a src folder in it.
The idea is not to have the package.json in the same folder as the source. That way I can easily symlink the src folder inside any app for fast dev.

After that I fixed a lot of issues. I also supported the full advertismentData.

The reason I am writing this issue is that I know a merge would be difficult. So I need to know if you are interested. If so we can look at how to do it. If not I will create a new npm package.

Thanks

EddyVerbruggen, 2018-07-16:
Hi, it would be great if we can find a way to merge your efforts into this repo. Can you isolate the important bits and send a PR? I'd be happy to make you a collaborator on this repo.

farfromrefug, 2018-07-16:
@EddyVerbruggen I tried to keep the commits well separated for that single reason. Will look what I can do. The commit that scares me is the one with the new directory structure.
Will keep you posted

EddyVerbruggen, 2018-07-16:
Yeah that one may hurt and is a non-(de facto)standard. Let's see how bad it is :)

farfromrefug, 2018-07-16:
@EddyVerbruggen don't worry. enjoy your vacations. don't have much time right now anyway. will let you know

allnash, 2018-11-29:
@farfromrefug when is the PR coming? :)

farfromrefug, 2018-11-30:
@allnash @EddyVerbruggen you are right long time coming. I did not have the time to do it and I was heaveliy working on my fork. it would / will be a huge merge request. I need to see with @EddyVerbruggen on how to do this. you can try my fork though as it is published. there are a lot of bug fixes and improvements. I am really hoping to find a way to merge this here. And thanks again @EddyVerbruggen for his work

allnash, 2018-11-30:
@EddyVerbruggen Any update?

allnash, 2018-11-30:
@farfromrefug I just installed your version. The documentations is a bit out of date? No?

farfromrefug, 2018-11-30:
@allnash might be. A very few api changed by name. What issue do you have?

allnash, 2018-11-30:
I am trying to write a bluetooth app, but I do not think NativeScript is behaving properly.
    ```CBCentralManagerDelegateImpl.prototype.centralManagerDidDiscoverPeripheralAdvertisementDataRSSI = function (central, peripheral, advData, RSSI) { }```
for example: in the above function, the advData param is empty for Eddystone beacon. I am feeling lost. Please advice.
I am pretty sure something is not right, has anyone else faced this issue?

farfromrefug, 2018-11-30:
@allnash advertisment should work in my fork. had issues with the original module. remember that sometimes the first event has no. adv for. a device. make sure you dont ignore next events as there might have been an update for your device with advertisment data. do you have a sample code?

bradmartin, 2018-11-30:
My fork also has advertisement working, which isn't too far off @farfromrefug fork. I've just added more BT functionality to do central and peripheral support.

allnash, 2018-11-30:
@farfromrefug Ha! You know that was the bug with this library. It only fired once.
@bradmartin let me try yours.

allnash, 2018-11-30:
@farfromrefug here is the code using your latest lib.

```
doStartScanning(): void {
        this.isLoading = true;
        this.peripherals.splice(0, this.peripherals.length);

        this._bluetooth.startScanning({
            // beware: the peripheral must advertise ALL these services
            filters: [{serviceUUID: "feaa"}],
            seconds: 10,
            onDiscovered: (peripheral) => {
                const distance = this.calculateDistance(peripheral.RSSI).toFixed(1);
                peripheral["distance"] = distance + "m away";
                console.debug(JSON.stringify(peripheral));
                this.peripherals.push(peripheral);
            },
            skipPermissionCheck: true
            // we can skip permissions as we use filters:
            // https://developer.android.com/guide/topics/connectivity/bluetooth-le
        }).then(() => {
                this.isLoading = false;
            },
            (err) => {
                this.isLoading = false;
                alert({
                    title: "Unable to scan.",
                    message: err,
                    okButtonText: "OK, got it"
                });
            });
}
```

allnash, 2018-11-30:
I tried to refer to AdvData but it is `{}`

allnash, 2018-11-30:
Here is the output log:
```
CONSOLE DEBUG file:///app/app/browse/browse.component.js:100:30: {"UUID":"AEAAE3EF-3152-FF78-2445-B024F178A0B5","name":"nRF5x","RSSI":-45,"advertismentData":{"uuids":{},"connectable":1},"state":"disconnected","rawAdvData":{},"distance":"0.1m away"}
CONSOLE LOG file:///app/tns_modules/nativescript-akylas-bluetooth/common.js:26:24: NativeScript-Bluetooth: INFO Bluetooth._isEnabled ---- this._centralManager.state: 5
CONSOLE LOG file:///app/tns_modules/nativescript-akylas-bluetooth/common.js:26:24: NativeScript-Bluetooth: INFO Bluetooth.startScanning ---- services: FEAA
CONSOLE LOG file:///app/tns_modules/nativescript-akylas-bluetooth/common.js:26:24: NativeScript-Bluetooth: INFO CBCentralManagerDelegateImpl.centralManagerDidDiscoverPeripheralAdvertisementDataRSSI ---- nRF5x @ -47 @ {
kCBAdvDataIsConnectable = 0;
kCBAdvDataServiceData =     {
FEAA = <10e8026f 742e6874 2f424f53 2d433030 31>;
};
kCBAdvDataServiceUUIDs =     (
FEAA
);
}
CONSOLE DEBUG file:///app/app/browse/browse.component.js:100:30: {"UUID":"AEAAE3EF-3152-FF78-2445-B024F178A0B5","name":"nRF5x","RSSI":-47,"advertismentData":{"uuids":{},"services":{}},"state":"disconnected","rawAdvData":{},"distance":"0.1m away"}
```

allnash, 2018-11-30:
@farfromrefug How can I get the raw data so I can build and Eddystone url frame from your API?

allnash, 2018-11-30:
Ok.

I made some progress in decoding it. I changed one file under iOS:
`CBCentralManagerDelegateImpl.ts`    `CBCentralManagerDelegateImpl.prototype.centralManagerDidDiscoverPeripheralAdvertisementDataRSSI = function (central, peripheral, advData, RSSI) { ... }`
```
 var rawData;
            if (advData.objectForKey(CBAdvertisementDataServiceDataKey)){
                var data = advData.objectForKey(CBAdvertisementDataServiceDataKey);
                var data = data.toString();
                rawData = data;
            };
            var payload = {
                UUID: peripheral.identifier.UUIDString,
                name: peripheral.name,
                localName: advertismentData['localName'],
                RSSI: RSSI,
                advertismentData: advertismentData,
                state: this._owner.get()._getState(peripheral.state),
                manufacturerId: manufacturerId,
                rawAdvData: rawData
            };
            this._owner.get()._advData[payload.UUID] = advertismentData;
            this._owner.get()._onDiscovered(payload);
            this._owner.get().sendEvent(ios_main_1.Bluetooth.device_discovered_event, payload)
```

farfromrefug, 2018-12-01:
@allnash This was already in here in my [fork](https://github.com/Akylas/nativescript-bluetooth/blob/master/src/ios/CBCentralManagerDelegateImpl.ts#L134). Now i see that i wrongly understood that key. I thought it was just a list of services but it is per service raw data.
I implemented better in this [commit](https://github.com/Akylas/nativescript-bluetooth/commit/e2cfdd48588e4bf8bdf869fc35bbd08135a70811)
And i publish 1.0.13 with this. You can try (though i can't test it)

allnash, 2018-12-01:
@farfromrefug : Unfortunately the data is not be sent upstream. Is there something going wrong in the NSData <--> JSON conversion?

farfromrefug, 2018-12-01:
The NSData is not converted to JSON per service. It s an arraybuffer per service. As we don't what s in it, it is yours to convert.

allnash, 2018-12-01:
Is the new version released?

farfromrefug, 2018-12-03:
@allnash Yes it is. BTW i enabled issues on my work. You should move that discussion there.
That way we can stay on track here and only talk about the merge

allnash, 2018-12-03:
@farfromrefug sounds good.

closed 2020-08-27 by farfromrefug