nativescript-community/ble #106: Get distance using RSSI value

ishaqueahmed-mev, 2018-10-23:
Hi,

I need to know how can we get the distance of the scanned devices or the connected device using the RSSi values?

Thanks.

allnash, 2018-11-29:
Here is the sample code @ishaqueahmed-mev 
```
calculateDistance(rssi): number {
       // you can pass txPower to the function
        const txPower = -59;

        if (rssi === 0) {
            return -1.0;
        }

        const ratio = rssi * 1.0 / txPower;
        if (ratio < 1.0) {
            return Math.pow(ratio, 10);
        } else {
            const distance = (0.89976) * Math.pow(ratio, 7.7095) + 0.111;
            // @ts-ignore
            return distance;
        }
    }
```

allnash, 2018-11-29:
Oh and then -> 
```
onDiscovered: (peripheral) => {
                    const distance = this.calculateDistance(peripheral.RSSI).toFixed(1);
                    peripheral["distance"] = distance + "m away";
                    console.debug(JSON.stringify(peripheral));
                    this.peripherals.push(<bluetooth.Peripheral>peripheral);
                },
```

ishaqueahmed-mev, 2019-09-25:
Thanks for the response, I had it done the same way you provided the method.

closed 2019-09-25 by ishaqueahmed-mev