nativescript-community/ble #109: tns preview cannot find this module

xinthose, 2018-11-11:
`tns preview`

**Error:**

> LOG from device XT1585: ERROR Error: Uncaught (in promise): Error: com.tns.NativeScriptException: Failed to find module: "nativescript-bluetooth", relative to: app/tns_modules/

I have run this command: `tns plugin add nativescript-bluetooth` and tried this one: `npm i nativescript-bluetooth --save-dev`.  I also added `"nativescript-bluetooth": "^1.3.1",` to `devDependencies` in file `package.json`.  

How do I include this plugin for testing?

Thank you.

EddyVerbruggen, 2018-11-11:
I don't think the Preview app ships with the Bluetooth plugin, but check this blog post: https://www.tjvantoll.com/2018/03/02/playground-plugins/

closed 2018-11-11 by EddyVerbruggen

xinthose, 2018-11-11:
Thank you Eddy.  [Apparently](https://docs.nativescript.org/tooling/docs-cli/project/testing/preview#command-limitations) this is a limitation of `tns preview`.