nativescript-community/ble #111: Release 2.0.0?

drewcovi, 2018-11-12:
At this point, examples and demos no longer work and the code is incompatible with Angular 6/Typescript...

Can we get a cut of 2.0.0?

bradmartin, 2018-11-12:
I forget what was needed during the overhaul of the plugin to publish it 🤔. It's possible it's already been done or tested since then. I've been using master as a base for an app for 6 months or so with no major bugs.

Until Eddy can publish it, you could fork, clone, `npm pack` it into your dependency for now. 

I have a local directory [here](https://github.com/PushTracker/EvalApp/blob/master/maxmobility/apps/eval-mobile/package.json#L61) that uses a local directory for the dependency. A different approach you could take until 2x is published.

drewcovi, 2018-11-12:
Thanks @bradmartin!

Seems I'm getting closer. Now I'm getting this when attempting imports.

```
Error: /Users/{user}/Development/{projectname}/node_modules/nativescript-bluetooth/common.ts is missing from the TypeScript compilation. Please make sure it is in your tsconfig via the 'files' or 'include' property.
The missing file seems to be part of a third party library. TS files in published libraries are often a sign of a badly packaged library. Please open an issue in the library repository to alert its author and ask them to package the library using the Angular Package Format
```

bradmartin, 2018-11-13:
ahhh okay, so make sure you transpile the source to .js using `tsc` (typescript compiler command). That should fix that issue and then you can `npm pack` it.

This is a different project but might help you figure out how to do this.

This [script cmd here](https://github.com/Max-Mobility/permobil-wear/blob/master/package.json#L19) is how I do it in a different project. Look at [this line](https://github.com/Max-Mobility/permobil-wear/blob/master/package.json#L29) for how to include the `.tgz` after packing 👍.

So with that in your project just execute it `npm run build-bluetooth` and it should all be good with a new `.tgz` each time you run it. This is assuming you're on Mac/linux since it uses `rm -rf` directly and not a windows compatible command for that. Just FYI 😄

drewcovi, 2018-11-13:
@bradmartin darn.. yeah I've set things up just as you have, but my project is typescript, so perhaps that's why it's still demanding *.ts files: 
```
node_modules/nativescript-bluetooth/bluetooth.ts is missing from the TypeScript compilation
```
Incidentally this only happens when I actually initialize Bluetooth. 

Thanks for your help with this, loving Nativescript thus far, and the community is great.

drewcovi, 2018-11-13:
@bradmartin ah! I just realized you're also using 1.3.0 not 2.0.0

ill dig and see if I can somehow get 1.3.0 to work.

drewcovi, 2018-11-13:
@bradmartin or is your copy not actually 1.3.0 but what you leveraged for your PR? It looks like everything is converted to typescript in your local repository.

bradmartin, 2018-11-13:
yea it's not 1.3.0 - I never changed the version after my fork 😉 - my local plugin has so much more bluetooth stuff in it and I want to PR it back into this but no time to really dig out the abstractions into a common API and actually test things out.

My fork was the base of the 2x TS refactor of this plugin, think that kicked off the overhauling of the plugin to TS. 

My projects also only use TypeScript and it works when I run those commands. You're definitely close to getting it, prob something silly happening that you'll 🤦‍♂️ when you get past it.

drewcovi, 2018-11-13:
yeah, id like to think im truly that close, but I've tried just about everything including symlinking bluetooth.ts to bluetooth.ios.ts (but then it just says it can't find bluetooth.ios.ts in the TypeScript compilation).

```
nativescript-bluetooth/src/bluetooth.ios.ts is missing from the TypeScript compilation. Please make sure it is in your tsconfig via the 'files' or 'include' property.
```

What's super annoying is I can't tell *which* tsconfig file needs updating. The tsconfig file in the plugin itself seems fine, and the tsconfig files in my project seem independent of node_modules:

```
{
  "compilerOptions": {
    "preserveSymlinks": true,
    "module": "commonjs",
    "skipLibCheck": true,
    "target": "es5",
    "pretty": true,
    "experimentalDecorators": true,
    "emitDecoratorMetadata": true,
    "noEmitHelpers": true,
    "noEmitOnError": false,
    "declaration": false,
    "lib": ["es6", "es2016", "dom"],
    "baseUrl": ".",
    "typeRoots": ["./node_modules/@types"]
  },
  "compileOnSave": false,
  "files": [
    "common.ts",
    "bluetooth.ios.ts",
    "bluetooth.android.ts",
    "./ios/ios_main.ts",
    "./ios/CBCentralManagerDelegateImpl.ts",
    "./ios/CBPeripheralDelegateImpl.ts",
    "./android/android_main.ts",
    "./android/TNS_BluetoothGattCallback.ts",
    "./android/TNS_LeScanCallback.ts",
    "./android/TNS_ScanCallback.ts"
  ],
  "exclude": ["node_modules", "platforms", "**/*.aot.ts"]
}
```

drewcovi, 2018-11-13:
hmmm upon further investigation i wonder if the fact im running nativescript-angular has anything to do with my issues...

bradmartin, 2018-11-13:
Shouldn't be an issue, one of the projects linked above I work on is also Angular.

Can you `cd` into the local directory and execute `tsc` without any errors?

drewcovi, 2018-11-13:
I can, so to ensure we're on the same page:

I can:
- Clone into my local Github/nativescript-bluetooth/src
- Run tsc
- Run npm pack
- Return to my project directory
- npm install {path to tgz}
```
npm notice 
npm notice 📦  nativescript-bluetooth@2.0.0
npm notice === Tarball Contents === 
npm notice 2.6kB  package.json                         
npm notice 2.2kB  base64.js                            
npm notice 481B   bluetooth.android.js                 
npm notice 290B   bluetooth.android.ts                 
npm notice 410B   bluetooth.ios.js                     
npm notice 223B   bluetooth.ios.ts                     
npm notice 6.6kB  common.js                            
npm notice 11.7kB common.ts                            
npm notice 8.5kB  index.d.ts                           
npm notice 236B   references.d.ts                      
npm notice 813B   tsconfig.json                        
npm notice 36.9kB android/android_main.js              
npm notice 30.9kB android/android_main.ts              
npm notice 14.9kB android/TNS_BluetoothGattCallback.js 
npm notice 13.7kB android/TNS_BluetoothGattCallback.ts 
npm notice 3.9kB  android/TNS_LeScanCallback.js        
npm notice 3.2kB  android/TNS_LeScanCallback.ts        
npm notice 5.3kB  android/TNS_ScanCallback.js          
npm notice 4.4kB  android/TNS_ScanCallback.ts          
npm notice 10.7kB ios/CBCentralManagerDelegateImpl.js  
npm notice 10.1kB ios/CBCentralManagerDelegateImpl.ts  
npm notice 15.0kB ios/CBPeripheralDelegateImpl.js      
npm notice 14.3kB ios/CBPeripheralDelegateImpl.ts      
npm notice 21.5kB ios/ios_main.js                      
npm notice 17.7kB ios/ios_main.ts                      
npm notice 509B   platforms/android/AndroidManifest.xml
npm notice 290B   platforms/ios/Info.plist             
npm notice 6.4MB  typings/android27.d.ts               
npm notice === Tarball Details === 
npm notice name:          nativescript-bluetooth                  
npm notice version:       2.0.0                                   
npm notice filename:      nativescript-bluetooth-2.0.0.tgz        
npm notice package size:  630.4 kB                                
npm notice unpacked size: 6.6 MB                                  
npm notice shasum:        726d69369c1e52c8145d61cd5760b8eddc0a3b4d
npm notice integrity:     sha512-rsJOTpwIifbN5[...]EJFXrxTZzmVNQ==
npm notice total files:   28                                      
npm notice 
nativescript-bluetooth-2.0.0.tgz
```

-

bradmartin, 2018-11-13:
maybe it's the ignore part :) - add this https://github.com/Max-Mobility/permobil-wear/blob/master/nativescript-bluetooth/.npmignore so when you `npm pack` it's not shipping the TS source and only the .js 

I think I ran into this also and forgot how I got over it, but that might be it ^^^

drewcovi, 2018-11-13:
### oh. em. gee.
[image]

drewcovi, 2018-11-13:
so yes, blocking ts from the bundle sealed the workaround deal. looks like this ticket is still valid until 2.0.0 can ship with built in ts files... not sure why it ran into those issues.

bradmartin, 2018-11-13:
The plugin still won't ship .ts source files since they're not executable by the JS engines. You only need the transpiled .js from the source TS files. 

I get what you're saying though, you want the 2x update 😄 👍

drewcovi, 2018-11-13:
Interesting... so any repo wouldn't be directly installable via Github? Or does `npm install` run a custom script which could include tsc?

bradmartin, 2018-11-13:
Nothing written in TS is executable, so yes you can't install it direct from github as TS. That's why libs written in TS have a dependency on TS so you can generate the .js to be executed.

`npm install` doesn't run anything special, that's where you could write your own `post-install` hook to handle that or a different script for users, but that's only from github or anywhere you don't have the .js files.

Again, typically, modules (packages) are published to NPM and you have the .js code to be executed. I don't include my TS files in published packages bc it's just a waste of bandwidth since it's not executable and you shouldn't be modifying anything inside of node_modules.

That is when you take this approach that we've done ^^^, fork from github, generate the .js source files and use a local instance as your dependency for that module.

Hope some of this makes sense :) wrapping up soon for the day so rushing through here.

closed 2020-08-27 by farfromrefug