nativescript-community/ble #114: onDiscovered never called

xinthose, 2018-11-24:
My code:
```
    constructor() {
        bluetooth.isBluetoothEnabled().then(
            function (enabled) {
                console.log("Enabled? " + enabled);
                if (enabled) {
                    bluetooth.startScanning({
                        serviceUUIDs: ["0100"],    // L2CAP --> 0100 (https://www.bluetooth.com/specifications/assigned-numbers/service-discovery)
                        seconds: 900,
                        onDiscovered: function (peripheral) {
                            console.log("Periperhal found with UUID: " + peripheral.UUID);
                        },
                        skipPermissionCheck: true,
                    }).then(function () {
                        console.log("scanning complete");
                    }, function (err) {
                        console.log("error while scanning: " + err);
                    });
                } else {
                    this.requestBluetooth();
                }
            }
        );
    }
```
I disconnect from my [Shutterbug](http://a.co/d/7zlEVFT) and reconnect but `onDiscovered` is never called.  The same happens with a blank `serviceUUIDs`.  I see `Enabled? true` in the log.

solev, 2019-04-23:
I have the same issue? Did you resolve it yet?
I am suspecting `skipPermissionCheck: true`. Have you requested permission before you start scanning?
I get an error when calling `requestCoarseLocationPermission()`

`ERROR Error: Uncaught (in promise): Error: java.lang.NullPointerException: Attempt to invoke virtual method 'int android.content.Context.checkPermission(java.lang.String, int, int)' on a null object reference
JS:     android.support.v4.content.ContextCompat.checkSelfPermission(ContextCompat.java:439)`

xinthose, 2019-04-23:
@solev Hello.  No, I did not.  I gave up on this project.  I'm not sure if I tried setting `skipPermissionCheck` to `false` or not.  The demo app did work for me though.

closed 2019-04-23 by xinthose