nativescript-community/ble #118: Is this repo still maintained?

drewcovi, 2018-12-20:
It appears that we have 7 pull requests, quite a few issues, and complete lack of gatt descriptor support at present. Is there a new recommended fork for this repository? Or can we get some pull requests updated?

EddyVerbruggen, 2018-12-20:
Please feel free to not ask that type of question on any OSS repo anywhere ever. Thank you.

I don't have infinite time and this plugin is simply not at the top of my pile of work. Perhaps next year. I also don't use it in a production app myself. Feel free to assist testing PRs, triaging issues, helping other users, etc.

drewcovi, 2018-12-20:
Not meant to draw ire. And I will certainly leave questions like this at any point in a repo that appears to be falling behind.

It’s okay to ask for help.

I’ve been in the same place many times myself. It helps to simply add additional people as maintainers.

And to be honest, there appear to be plenty of willing contributors.