nativescript-community/ble #119: iOS: not storing strong ref to peripheral causes API MISUSE error

rynop, 2019-01-08:
When calling `connect()` I sometimes (like 1 out of 10 times) get this error:

> API MISUSE: Cancelling connection for unused peripheral <private>, Did you forget to keep a reference to it?

I've never written C#/Swift, but from searching, it seems as if the iOS native code is not storing a **strong** reference to the peripheral's that the last `scan()` had discovered.  Here is a SO solution to the problem:
https://stackoverflow.com/questions/34837148/error-corebluetooth-api-misuse-cancelling-connection-for-unused-peripheral

Has anyone seen this? Someone that knows iOS dev, is this a simple fix that can be made in this plugin?  Is it related to https://github.com/EddyVerbruggen/nativescript-bluetooth/issues/93 ?

FWIW I'm using the [v2 branch](https://github.com/EddyVerbruggen/nativescript-bluetooth/tree/v2) of this repo.

Here is my pseudo code (added some comments inline):

```typescript
...
    const peripheral = await new Promise<Peripheral>((resolve, reject) => {
      // The next line is where I intermittently get the API MISUSE error
      this.bluetooth.connect({
        UUID: perfUuid,
        onConnected: data => {
          const p: Peripheral = { ...data, RSSI: 0 };
          resolve(p);
        },
        onDisconnected: p => {
          console.log('Periperhal disconnected with UUID: ' + p.UUID);
          reject(
            `Periperhal ${perfUuid} disconnected before read could be made`
          );
        }
      });
    });

    console.log('Connected to: ', peripheral.UUID);

    for (let i = 0; i < 10; i++) {
      const readRes = await this.bluetooth.read({
        peripheralUUID: perfUuid,
        serviceUUID: BLE_SERVICE_ID,
        characteristicUUID: MY_BLE_CHARACTERISTIC
      });
      const proximity = Buffer.from(readRes.value).readUInt16LE(0, true);
      console.log('sample', proximity);
      samples.push(proximity);

      this.captureProgress += 10;
      await asleep(200);
    }

    // If I don't disconnect (comment out lines below) I seem to get the API MISUSE error less...
    try {
      await this.bluetooth.disconnect({ UUID: perfUuid });
    } catch (error) {
      console.error(`Failed to disconnect ${perfUuid}`);
    }
```

farfromrefug, 2019-01-09:
I think i know what's going on. I don't see it in your code but are you sure you don't start a new scanning just before connecting? On new scan i clear all discovered peripheral references 
Thus if you start a new scan and immediatly connect to a peripheral discovered previously (and not yet discovered again) you might end up with that issue.

rynop, 2019-01-09:
Thanks Martin. Here is how my logic goes:

1.  Scan, display discovered peripherals in `<ListView>`.  I filter based on a known service UUID. So I only get a few peripherals.
1.  On `tap()` open modal.  In `ngAfterViewInit` of component displayed in modal, I sleep for 200 ms, then do a `.connect()`

The amount of time between the `scan()` and `connect()` are variant (because end user could take as long as they want before `tap()`ing on peripheral) but minimum is 200ms.

Is there a minimum time you recommend between doing a `scan()` and `connect()`? I can just increase my sleep.

I could work-around this issue, however I can't seem to find a way to catch the `API MISUSE` error so that I can recover from it.  Here is my code - the `.catch()` nor the `catch (error)` are ever hit.  How can I catch `API MISUSE` exceptions?

```typescript
  private async bleConnect(retry?: boolean): Promise<Peripheral> {
    await asleep(200);
    const perfUuid = this.metaData.id;
    try {
      const peripheral = await new Promise<Peripheral>((resolve, reject) => {
        this.bluetooth
          .connect({
            UUID: perfUuid,
            onConnected: data => {
              const p: Peripheral = { ...data, RSSI: 0 };
              resolve(p);
            },
            onDisconnected: p => {
              console.log('Periperhal disconnected with UUID: ' + p.UUID);
              reject(
                `Periperhal ${perfUuid} disconnected before read could be made`
              );
            }
          })
          .catch(e => {
            // I'm never hit on `API MISUSE` error
            console.error('Problem connecting', e);
            reject(e);
          });
      });
      return peripheral;
    } catch (error) {
      // I'm never hit on `API MISUSE` error
      if (!retry) {
        console.error('Problem connecting, retrying..');
        await asleep(200); // @see https://github.com/EddyVerbruggen/nativescript-bluetooth/issues/119
        return await this.bleConnect(true);
      } else {
        throw error;
      }
    }
  }
```

And lastly, I think a pretty common use case is to scan then connect shortly after.  It would be cool if the plugin handled it.  Ex: On your personal phone, you scan for you BLE audio earbuds and want to connect right away.

Wish I could help fix, I just don't have time at the moment to learn iOS & Android native dev.

farfromrefug, 2019-01-09:
Do you call multiple scans in a raw?

rynop, 2019-01-09:
I allow the end user to click a 're-scan` button, but I'm pretty sure I hit the `API MISUSE` error when only scanning once. I'm trying to verify this now.

I do have a use case where I need to auto-scan every X seconds.  As the end user is walking around using my app I want to automatically pick up any new peripherals that they come in close proximity to.  To be clear though, I haven't started testing this code path yet.  I hit the `API MISUSE` error above without doing an auto re-scan.  My hunch, given what you are saying, is I will hit the error much more frequently when I start testing this use case.

The popular [LightBlue Explorer](https://punchthrough.com/) app (bottom of page), does the auto rescan and allows instant connection, so I know from a technology perspective its possible to accomplish.

farfromrefug, 2019-01-11:
I have looked at this and it fact it has nothing to do with scan peripherals anymore. I now ask the centralManager for the peripheral before connecting to it. The error is somewhere else.
Can you get a log (bluetooth debug) so that i can see exactly when this happens?

rynop, 2019-01-11:
Thanks for looking.  

Sorry for my ignorance, but can you point me to directions on how to enable "bluetooth debug"?  I'm guessing it is not a nativescript or `nativescript-bluetooth` plugin setting.  Is it the `Bluetooth` profile/log instructions [here](https://developer.apple.com/bug-reporting/profiles-and-logs/)?

Unfortunately, I'm not going to be near a computer for the rest of the day or this weekend, so I won't be able to provide logs until next week.

I've also added you to my private repo that contains the app where I'm seeing the issue.  I'm not doing this because I want to exploit you to fix my bugs :) I'm just doing this in case it helps you track down what may be going on.  Feel free to look at it if you like - or completely ignore.  Please don't share with others - you are an OSS dev that gives back to the community so I trust you ;) If you do choose to look, I connect to proprietary BLE peripherals, so you won't be able to replicate.  I'm able to hit the issue if I hit the `Admin` tab at the bottom, click on the peripheral (which opens `calibrateXX.component`).  I then click on the `Capture` button (which does the `read()`s.  If I close this modal and re-do these steps (reopen, capture etc), it will sometimes throw the error.

Please don't spend anytime fixing my code, I don't want to waste your time.  My intent here is maybe if you have access to a codebase that is exhibiting the issue it may help you track it down quicker..

closed 2020-12-10 by farfromrefug