nativescript-community/ble #121: v2 release

rynop, 2019-01-09:
Starting a list here of things that need to be included in the v2 release before the [v2 branch](https://github.com/EddyVerbruggen/nativescript-bluetooth/tree/v2) can be merged into `master`.  The v2 branch is primed from the https://github.com/Akylas/nativescript-bluetooth fork.  See https://github.com/Akylas/nativescript-bluetooth/issues/5 for history.

My hope is we can discuss at a high level what needs to be in v2, then make an issue for each feature adding it to the [2.0.0 milestone](https://github.com/EddyVerbruggen/nativescript-bluetooth/milestone/4).  In the individual feature issues we can have lower level discussion if needed.

- [ ] [Migration/upgrade guide](https://github.com/EddyVerbruggen/nativescript-bluetooth/issues/120)
- [ ] Comb through README.md to make sure features/changes in `v2` branch are documented.
- [ ] [demo-ng](https://github.com/EddyVerbruggen/nativescript-bluetooth/tree/v2/demo-ng) ts overhaul (was written long ago before features like `async/await`)
- [ ] Update [demo](https://github.com/EddyVerbruggen/nativescript-bluetooth/tree/v2/demo) with features/changes that are included in `2.0.0`

@EddyVerbruggen, @farfromrefug, and others, feel free to make suggestions below and/or tag existing issues with `v2` and add em to the `2.0.0` milestone.

farfromrefug, 2019-01-24:
2.0.0-beta.1 released. Already using it in prod

phil-sch, 2019-03-12:
Hi @farfromrefug, could you provide a brief breakdown on how to manually import this branch of the plug-in into an existing project?

I already tried copying the downloaded `nativescript-bluetooth-2` folder into `node_modules` and using it via `var bluetooth = require("nativescript-bluetooth-2");` but I get an error saying:

> Cannot find module 'nativescript-bluetooth-2'

I'm mainly interested in using the `filters` functionality of `startScanning`, since that feature seems to be broken in the master branch.

farfromrefug, 2019-03-12:
@phil-sch no need use special branch. Simply use the beta tag!
Look [here](https://www.npmjs.com/package/nativescript-bluetooth) i am at 2.0.0-beta.11

farfromrefug, 2019-04-18:
@EddyVerbruggen What do you think about making v2 official?
I really think beta 14 is a really good candidate.

EddyVerbruggen, 2019-04-18:
@farfromrefug Happy to do so! Anyone here who doesn't think that's a good idea?

phil-sch, 2019-04-18:
I've been using the latest beta versions for some time now and they work like a charm. 💙

farfromrefug, 2019-04-18:
@EddyVerbruggen BTW we would need the help in getting the documention updated. Honestly not really interested by docs :s

erikbjork, 2019-04-26:
I have found two bugs in 2.0.0-beta.12 which I think still exists in the latest beta (2.0.0-beta.14). They're both in the Android implementation. One is a failure when a scanned device has no device name and the other is a problem with `this.connection` map containing null values. I have working patches for the 2.0.0-beta.12 release.

I will confirm that they still exists in the latest beta and, if so, get back with better descriptions of the bugs. Do you want me to post them as separate issues?

farfromrefug, 2019-04-26:
@erikbjork  let s handle them here. a walkthrough on how to reproduce them would be awesome. maybe even a nodejs  bleno  example to simulate a device if it depends on it. need to check but think I created one in the v2  branch

erikbjork, 2019-04-26:
I could not run my application with the 2.0.0-beta.14 release (because of es2015 import issues). But @farfromrefug has got the details and will make some changes.

farfromrefug, 2019-05-31:
@erikbjork issue is fixed right now. Anyone to help a bit on the doc?

farfromrefug, 2019-07-20:
@EddyVerbruggen I am starting to think i should create a new plugin based on v2. The reason is the issue people have with understanding that that plugin is for BLE devices.
So i was thinking about creating ```nativescript-ble``` out of the v2 branch.
And in the future maybe this plugin could become the plugin for Bluetooth 2.0. Dont want to create the new plugin without brainstorming with you

EddyVerbruggen, 2019-07-20:
Hi Martin!

I'd rather not have another Bluetooth plugin out there. Maybe we can merge v2 to master and publish it to npm if you think it's time to do so.

Then if Bluetooth Classic can/will be added we can publish another major version to npm and remove the "this is for BLE only" stuff from the readme.

farfromrefug, 2019-07-23:
@EddyVerbruggen To be honest seeing the structure of the plugin (static object handling everything) it wont be easy to structure it to handle both BLE and non BLE.
V2 is fully ready to be published for {N}6. Surely still some little bugs but nothing major.

However i am still really thinking about creating a new plugin. Either 2 plugins one for BLE and one for non BLE as those devices are really handled differently. Or use only one plugin with a totally different layout, like i did in Titanium:  No Static object, Bluetooth and BLE object class. That way you can have shared code and different intelligence.

Maybe we can talk about this over slack. Would love to have your input

closed 2020-08-27 by farfromrefug