nativescript-community/ble #122: V2 Bluetooth.Write Changes

JakeAi, 2019-01-10:
I used to be able to write a buffer to the write function like

```TS
let converter    = createConverter( struct( {
    setRPM  : uint32,
    savedRPM: uint32,
    persist : uint32,
} ), <DecoderOptions>{ validate: false } );

let encodeResult = converter.encode( {
    setRPM  : rpm,
    savedRPM: 0,
    persist : ( !persist ? 0 : 1 ),
} );

 return await bluetooth.write( <WriteOptions>{
    peripheralUUID    : this.uuid,
    serviceUUID       : GATT.uuid,
    characteristicUUID: GATT.Characteristics.RPM.uuid,
    value             : encodeResult,
} );
```

Now it is completely broken because nativescript-bluetooth takes my value, which is a Buffer, and converts it to a Uint8Array? Not really sure what's going on. Everything I've tried Buffer related fails with different `Invalid Value` errors. @rynop seeing as you've been pretty active, do you know how to fix this? I don't want to do anything hacky as it was fine before. I am sending it to a BLE chip on our own board and it expects a struct.

JakeAi, 2019-01-10:
I believe the problem lies between 
https://github.com/EddyVerbruggen/nativescript-bluetooth/blob/master/src/ios/ios_main.ts#L507
and
https://github.com/EddyVerbruggen/nativescript-bluetooth/blob/v2/src/bluetooth.ios.ts#L629

V2 checks for Array.isArray and all Buffers and UintArrays are not "isArray" so it will fail. I was using binary data in the original version.

farfromrefug, 2019-01-10:
@JakeAi that should be fairly easy to fix. Can you share the code for ```createConverter```?

JakeAi, 2019-01-10:
@farfromrefug createConverter comes from https://github.com/maxdavidson/structly. It's an amazing package!

farfromrefug, 2019-01-13:
@JakeAi should be fixed now

JakeAi, 2019-01-14:
@farfromrefug how are you debugging this? When I try and debug/build the project I get 
```
*** Terminating app due to uncaught exception 'NativeScript encountered a fatal error: Error: Could not find module '../bluetooth.ios'. Computed path '/var/mobile/Containers/Data/Application/9CB0BBCC-924F-459B-8910-33C8D97A21BA/Library/Application Support/LiveSync/app/tns_modules/nativescript-bluetooth/bluetooth.ios'.

```

farfromrefug, 2019-01-14:
what project are you talking about? yours or the demo? if you are using the demo, you need to run ```npm run tsc``` in the plugin root folder first

JakeAi, 2019-01-14:
The demo. I cloned it on my Macbook. Ran `npm install` then `npm run tsc` then `npm run demo.ios`. At runtime I get the error below. When I check the modules folder on the app the bluetooth.ios and bluetooth.android files are not there. 

The first image is the plugin folder in the build.
[https://imgur.com/1lzoy9v](https://imgur.com/1lzoy9v)

This image is the folder that is bundled in the app. The files don't get copied.
[https://imgur.com/QH5fgmY](https://imgur.com/QH5fgmY)

```
JavaScript error:
file:///app/tns_modules/nativescript-bluetooth/ios/CBCentralManagerDelegateImpl.js:4:30: JS ERROR Error: Could not find module '../bluetooth.ios'. Computed path '/Users/jake/Library/Developer/CoreSimulator/Devices/F0AD12EE-4483-4678-B015-4DFB909B24DA/data/Containers/Bundle/Application/168BDEDD-88BC-4F06-B664-A87822117C2A/demo.app/app/tns_modules/nativescript-bluetooth/bluetooth.ios'.
(CoreFoundation) *** Terminating app due to uncaught exception 'NativeScript encountered a fatal error: Error: Could not find module '../bluetooth.ios'. Computed path '/Users/jake/Library/Developer/CoreSimulator/Devices/F0AD12EE-4483-4678-B015-4DFB909B24DA/data/Containers/Bundle/Application/168BDEDD-88BC-4F06-B664-A87822117C2A/demo.app/app/tns_modules/nativescript-bluetooth/bluetooth.ios'.
at
1   require@file:///app/tns_modules/nativescript-bluetooth/ios/CBCentralManagerDelegateImpl.js:4:30
2   anonymous@file:///app/tns_modules/nativescript-bluetooth/ios/CBCentralManagerDelegateImpl.js:4:30
3   evaluate@[native code]
4   moduleEvaluation@[native code]
5   promiseReactionJob@[native code]
6   require@[native code]
7   anonymous@file:///app/tns_modules/nativescript-bluetooth/bluetooth.js:4:45
8   evaluate@[native code]
9   moduleEvaluation@[native code]
10  promiseReactionJob@[native code]
11  require@[native code]
12  anonymous@file:///app/main-view-model.js:7:39
13  evaluate@[native code]
14  moduleEvaluation@[nati<…>
^CExecuting after-watch hook from /Users/jake/Downloads/nativescript-bluetooth-2/demo/hooks/after-watch/nativescript-dev-typescript.js
Stopping tsc watch
Executing after-watch hook from /Users/jake/Downloads/nativescript-bluetooth-2/demo/hooks/after-watch/nativescript-dev-webpack.js
Stopping webpack watch
Executing after-watch hook from /Users/jake/Downloads/nativescript-bluetooth-2/demo/hooks/after-watch/nativescript-dev-typescript.js
Executing after-watch hook from /Users/jake/Downloads/nativescript-bluetooth-2/demo/hooks/after-watch/nativescript-dev-webpack.js
Stopping webpack watch

```

farfromrefug, 2019-01-15:
@JakeAi can you try again? Made a few changes. Tested on iOS now and it works fine.

JakeAi, 2019-01-15:
@farfromrefug i'm getting the same error. Can you clone a fresh pull and try it? My buddy is getting the same issue. I've tried it on two computers. I also tried `tns plugin add ../../nativescript-bluetooth` in my own project and I still get the same issue. `JS ERROR Error: Could not find module '../bluetooth.ios'.`

farfromrefug, 2019-01-16:
should be fxed . tried myself on iOS and it works fine.

JakeAi, 2019-01-16:
So it's a webpack issue? Is that my fault or the the plugin structure's fault?

farfromrefug, 2019-01-16:
no it s a main issue which does not appear with web pack. and I was using web pack... should be good now

farfromrefug, 2019-01-24:
@JakeAi is it fixed?

JakeAi, 2019-01-24:
The building is fixed yes. The writes—I'm not 100% sure. I think the buffer wasn't working still. But I have been rewriting this plugin to be async await and a lot of other minor fixes. Expect a PR in the near future! :)

JakeAi, 2019-01-24:
@farfromrefug Fork is here https://github.com/JakeAi/nativescript-bluetooth/tree/v2/src

farfromrefug, 2019-01-24:
@JakeAi  please could you make separated PRs for your changes? I will take most of them. only the refactoring I am not sure about. tslint is used to format the code. your changes breaks tslint. thanks

JakeAi, 2019-02-01:
@farfromrefug yes I can, currently I'm using it in a project, so I can't necessarily take the time to make separate PRs, but once I am done I will. What part of refactoring are you unsure about?

farfromrefug, 2019-02-01:
@JakeAi sure thanks! BTW you will have to merge and it might not be fun :s Kind of change a lot of things!

JakeAi, 2019-02-01:
@farfromrefug is there a reason you aren't using async await?

farfromrefug, 2019-02-01:
@JakeAi wasn't native! promise are

JakeAi, 2019-02-01:
@farfromrefug well the plugin is targeting ES5, so technically, it would be ok to use..? lol

farfromrefug, 2019-02-01:
definitely OK to use. just need to look exactly how ts transpile  it before I get ready to use them. plus i am so used to promise

JakeAi, 2019-02-01:
If you look at my fork, most of the iOS stuff is async await. Should be a pretty solid template.

closed 2020-08-27 by farfromrefug