nativescript-community/ble #123: Unable to detect mobile phones and laptops

Suralal001, 2019-01-11:
The demo-ng app is not able to detect mobile devices and laptops. but I could find that it was able to discover smartwatch Bluetooth.

farfromrefug, 2019-01-13:
@Suralal001 I think it s normal. An't those bluetooth 2.0 devices? That module is for BLE devices only

suralal, 2019-01-13:
Is there any possible way to connect with a raspberry pi.i am not sure with the Bluetooth version of it

farfromrefug, 2019-01-13:
Yes you can easily create a BLE device on Pi. Look at bleno (there s an example [here](https://github.com/EddyVerbruggen/nativescript-bluetooth/tree/v2/ble_server_test))

suralal, 2019-01-13:
I wanted to transfer data from mobile app to pi using the plugin.hope that would work

farfromrefug, 2019-01-13:
Yes you can. I am closing that issue as it is getting out of topic. Please use Slack for those kind of questions

closed 2019-01-13 by farfromrefug